uriyacovy/ESPHome_nuki_lock #5: Feature request: Open doors with door knobs is missing.

shiSHARK, 2022-08-12:
The integration works great. 

But I'm missing the open door feature. Since my door is with a door knob and not with a handle on the outside I can't open the door from the outside. It will be great to add open functionality as a separate entity.

I really like that it allows me to unlock and lock the door without actually opening it. But there is need for open door button where the outside of the door is using a knob.

uriyacovy, 2022-08-13:
Hi shiSHARK,
This is already implemented with HA open service. See here: https://github.com/uriyacovy/ESPHome_nuki_lock/issues/1
Let me know if this doesn't work for you.

crixionz, 2022-08-16:
The issue here might be that HA's frontend doesn't show an open button (yet). 
I filed this [esphome issue #3495](https://github.com/esphome/issues/issues/3495) and they confirmed it's an issue of HA, not ESPHome.

@shiSHARK try this in [🔨Developer Tools → Services](https://my.home-assistant.io/redirect/developer_services/):
```yaml
service: lock.open
target:
  entity_id: lock.mynuki   (← or whatever your nuki's entity_id is in your HA) 
  ```
It should open your nuki door. 
If that works, you can use it in automations or scripts.
Or you could set up a button in ESPHome which you can then use in Lovelace:
```yaml
esphome: 
# ...
button:
  - platform: template
    name: "Nuki Open"
    on_press:
      - lock.open: mynuki

lock:
  - platform: nuki_lock
    id: mynuki
    # ...
```

uriyacovy, 2022-08-21:
Thanks @crixionz. Did you open an issue on HA?
@shiSHARK, waiting also for your feedback.

crixionz, 2022-08-21:
I did not (yet) @uriyacovy 😐 I want to open a different issue for your component first but am currently procrastinating 😬

uriyacovy, 2022-08-21:
Well, I'm on my summer vacation anyway, so take your time :smile:.

uriyacovy, 2022-11-25:
Close due to inactivity.

closed 2022-11-25 by uriyacovy