uriyacovy/ESPHome_nuki_lock #11: State won't update when turning by hand

nikiwiener, 2022-11-15:
When turning the nuki lock by hand, the locked/unlocked state won't update.

uriyacovy, 2022-11-15:
Can you specify the HW (nuki and esp) and the yaml you use?

drdino, 2022-11-15:
@nikiwiener i believe manual usage of the lock (spinning the wheel manually) is not supported generally (also not by Nuki).

nikiwiener, 2022-11-15:
I'm using the Smart Lock 3.0 (not Pro) and a ESP32 NodeMCU board (There's "ESP32-WROOM-32" written on the ESP chip)

yaml file:

esphome:
  name: esp-nuki-bridge
  libraries:
    - Preferences
    - https://github.com/uriyacovy/NukiBleEsp32#dev

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@dev

esp32:
  board: "esp32dev"  
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected:
      name: "Nuki Connected"
    is_paired:
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXX"

ota:
  password: "XXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Nuki-Bridge Fallback Hotspot"
    password: "XXXX"

captive_portal:

nikiwiener, 2022-11-15:
> @nikiwiener i believe manual usage of the lock (spinning the wheel manually) is not supported generally (also not by Nuki).

Are you sure? It updates in homekit + the nuki app as expected.

haseat, 2022-11-15:
> > @nikiwiener i believe manual usage of the lock (spinning the wheel manually) is not supported generally (also not by Nuki).
> 
> Are you sure? It updates in homekit + the nuki app as expected.

I just tried, it works fine for me

nikiwiener, 2022-11-16:
> > > > @nikiwiener i believe manual usage of the lock (spinning the wheel manually) is not supported generally (also not by Nuki).
> > > 
> > > 
> > > Are you sure? It updates in homekit + the nuki app as expected.
> > 
> > 
> > I just tried, it works fine for me
> 
> It works for me as well (even though it is a use case I never tested).
> 
> @nikiwiener, I am curious, what is the scenario that leads to the manual turning anyway?

I'm using it to lock/unlock when I'm inside, since it's faster/ more quiet.

nikiwiener, 2022-11-16:
Did a restart of the esp today & now it updates.

closed 2022-11-16 by nikiwiener