uriyacovy/ESPHome_nuki_lock #12: nvs_get_blob len fail: bleAddress NOT_FOUND

zvucetic, 2022-11-22:
HI,

I'm using ESPHome version  2022.11.1 

I'm getting this issue (message) in the logs and Bluetooth is not working, not sure if the device is faulty or some other problem:

[04:00:08][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:09][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:09][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:10][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:10][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:11][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:11][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:12][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:12][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[04:00:13][E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND

[logs_nuki-bridge_upload (1).txt](https://github.com/uriyacovy/ESPHome_nuki_lock/files/10063137/logs_nuki-bridge_upload.1.txt)

uriyacovy, 2022-11-25:
It looks like the module is not paired to Nuki. Did you follow the pairing sequence ([here](https://github.com/uriyacovy/ESPHome_nuki_lock#how-to-use))?

zvucetic, 2022-11-25:
From what I managed to find out seems it's related to the ESP32 module. I ordered a new one, slightly different model, so I will report once it's comes.

uriyacovy, 2022-11-25:
OK, please update if the problem is solved.

zvucetic, 2022-12-05:
Hi, 
with the new ESP32 it works, so it was the HW issue. 
Thanks for the support.

FelixAhrens, 2022-12-29:
@zvucetic I have the same problem, with 3 different boards, not sure if it's the HW. Which board works for you?

When I try to connect anyway, I get the following:

```
[ 70195][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
connectBle retry 0
pairNuki: connectBle success after 1050 ms
sendPlainMessage: connectBle success after 0 ms
E (79122) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (79122) task_wdt:  - loopTask (CPU 1)
E (79122) task_wdt: Tasks currently running:
E (79122) task_wdt: CPU 0: IDLE
E (79122) task_wdt: CPU 1: IDLE
E (79122) task_wdt: Aborting.

abort() was called at PC [...] on core 0

Backtrace: [...] |<-CORRUPTED
```
Can anyone help?

Thanks!

uriyacovy, 2022-12-31:
Can you post your yaml?

FelixAhrens, 2022-12-31:
```
esphome:
  name: nuki-bridge

  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32#dev

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@dev

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0

lock:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: [...]

ota:
  password: [...]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nuki-Bridge Fallback Hotspot"
    password: [...]

captive_portal:
    
```

uriyacovy, 2022-12-31:
Can you try without encryption?

FelixAhrens, 2022-12-31:
same log with

```
api:
#  encryption:
#    key:
```

ruimarinho, 2023-01-02:
I'm having the same issue on the dev branch with a lolin32.

uriyacovy, 2023-01-02:
Can you check if the

FelixAhrens, 2023-01-03:
Thanks for your effort! Unfortunately not in esphome, I will test it with arduino, but that have to wait until the weekend.

Which esp32 board do you use? I will also test more HW...

zvucetic, 2023-01-03:
Hi,

The only two versions I managed to make work are both with the external antenna. 

Maybe the problem is signal (even thou I put the ESP on the NUKI and it's still not working)

I also bought this AZDelivery from Amazon (second), not working 🤦

zvucetic, 2023-01-04:
There are lot's of different models and manufacturers, this two (along AZDelivery are also not working

[image]
[image]
[image]

FelixAhrens, 2023-01-05:
thanks for the board infos. a cheap esp cam is also not working for me, the days i get a D1, maybe this will work.

the wireless logs are a little bit different (same via web.esphome.io and usb -> bleAddress NOT_FOUND). 

```
[17:40:20][VV][scheduler:196]: Running interval 'update' with interval=500 last_execution=102332 (now=102832)
INFO nuki-bridge.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO nuki-bridge.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for nuki-bridge.local
WARNING Disconnected from API
```

the brigde is conncted to home assistant but not paired (Running interval 'update'...). if I set the nuki lock in pairing mode, the brigde is disconnecting. same with and without encryption.

Jejebond, 2023-01-05:
Same issue here "[ 42837][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND" with AZdelivery V4

zvucetic, 2023-01-14:
I tried using the same ESP32 for a different purpose to use with iTag buttons, but it will not discover them. 
Probably it's the same issue as why it's not working with NUKI. 
Probably it's the chip not finding exactly the channel we need for NUKI.  
At first I taught it will not find any BLE, but it will find the Xiaomi temperature sensor and Mi Flora sensors. 
Strange things... :|

uriyacovy, 2023-01-15:
@zvucetic, Can you check if the [NukiBleEsp32](https://github.com/I-Connect/NukiBleEsp32) sample works for you?

Jejebond, 2023-01-15:
> I tried using the same ESP32 for a different purpose to use with iTag buttons, but it will not discover them. Probably it's the same issue as why it's not working with NUKI. Probably it's the chip [...]

I've used my Azdelivery V4 ESP for an other integration (Switchbot) and it works flawlessly.

FelixAhrens, 2023-01-15:
maybe there where changes in the preferences library?

zvucetic, 2023-01-16:
> @zvucetic, Can you check if the [NukiBleEsp32](https://github.com/I-Connect/NukiBleEsp32) sample works for you?

How do I flash that?
I tried this https://github.com/technyon/nuki_hub and it doesn't work. 
Even if I start BLE scanner, it will not see Nuki (and some other BLE devices like iTag) so I suspect it's something HW related.

uriyacovy, 2023-01-29:
nuki_hub uses the same NukiBleEsp32 library. If this does not work, I suggest to open an issue there.

zvucetic, 2023-02-11:
Hi,

this device is working normally
https://www.amazon.de/dp/B0B55MS33L?psc=1&ref=ppx_yo2ov_dt_b_product_details

uriyacovy, 2023-02-11:
Great to hear. 
What's the different between this and the ESP you used previously?

FelixAhrens, 2023-02-13:
> Hi,
> 
> this device is working normally https://www.amazon.de/dp/B0B55MS33L?psc=1&ref=ppx_yo2ov_dt_b_product_details

@zvucetic can you share your yaml code? Are you also on ESPHome version 2022.12.8? I got the same board but still same error..

zvucetic, 2023-02-15:
> Great to hear. What's the different between this and the ESP you used previously?

Don't know, but seems it's hardware related. 

-- My YAML is no different from the stock, I'm using DEV enviroment. 
I'm on : 2022.12.8 version

esphome:
  name: "48-esp32-nuki"

  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32#dev

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@dev

esp32:
  board: esp32dev
  framework:
    type: arduino
#    version: 2.0.3
#    platform_version: 4.4.0

button:
  - platform: template
    name: "Nuki Open"
    on_press:
      - lock.open: mynuki

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    id: mynuki
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"

FelixAhrens, 2023-02-18:
Strange. Same board (also tried 5 others), same code, different result (or same, the error code above). Someone any idea?

uriyacovy, 2023-04-27:
Closed due to inactivity.

closed 2023-04-27 by uriyacovy