uriyacovy/ESPHome_nuki_lock #15: Installation no longer works since ESPHome version v2022.12.0

axute, 2022-12-14:
```
INFO Reading configuration /config/esphome/esp32-nuki.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-nuki (board: az-delivery-devkit-v4; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing git+https://github.com/nkolban/ESP32_BLE_Arduino/
INFO Installing git+https://github.com/nkolban/ESP32_BLE_Arduino/
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-wszsbpf1'...
Library Manager: ESP32 BLE Arduino@1.0.1+sha.adc2aee has been installed!
INFO ESP32 BLE Arduino@1.0.1+sha.adc2aee has been installed!
Library Manager: Installing git+https://github.com/uriyacovy/NukiBleEsp32
INFO Installing git+https://github.com/uriyacovy/NukiBleEsp32
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-k0s41b7s'...
Library Manager: Nuki Ble for Esp32@0.0.4+sha.b0f8588 has been installed!
INFO Nuki Ble for Esp32@0.0.4+sha.b0f8588 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing git+https://github.com/h2zero/NimBLE-Arduino.git
INFO Installing git+https://github.com/h2zero/NimBLE-Arduino.git
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-3sf5trf5'...
Library Manager: NimBLE-Arduino@1.4.1+sha.0d9f039 has been installed!
INFO NimBLE-Arduino@1.4.1+sha.0d9f039 has been installed!
Library Manager: Installing git+https://github.com/vinmenn/Crc16.git
INFO Installing git+https://github.com/vinmenn/Crc16.git
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-jtx9t_ac'...
Library Manager: CRC16@0.0.0+20221214123502.sha.f46d86b has been installed!
INFO CRC16@0.0.0+20221214123502.sha.f46d86b has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing Syslog @ 2.0.0
Downloading 0% 10% 20%

Unpacking  [------------------------------------]    0%
Unpacking  [##----------------------------------]    6%
Unpacking  [####--------------------------------]   12%
Unpacking  [######------------------------------]   18%
Unpacking  [#########---------------------------]   25%
Unpacking  [###########-------------------------]   31%
Unpacking  [#############-----------------------]   37%
Unpacking  [###############---------------------]   43%
Unpacking  [##################------------------]   50%
Unpacking  [####################----------------]   56%
Unpacking  [######################--------------]   62%
Unpacking  [########################------------]   68%
Unpacking  [###########################---------]   75%
Unpacking  [#############################-------]   81%
Unpacking  [###############################-----]   87%
Unpacking  [#################################---]   93%
Unpacking  [####################################]  100%
Library Manager: Syslog@2.0.0 has been installed!
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- Preferences @ 2.0.0
|-- ESP32 BLE Arduino @ 1.0.1+sha.adc2aee
|-- Nuki Ble for Esp32 @ 0.0.4+sha.b0f8588
|   |-- NimBLE-Arduino @ 1.4.1+sha.0d9f039
|   |-- CRC16 @ 0.0.0+20221214123502.sha.f46d86b
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Syslog @ 2.0.0
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/api_connection.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/api_server.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/list_entities.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/proto.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/api/user_services.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/button/button.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/esp32/core.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/esp32/gpio.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/lock/lock.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/logger/logger.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/md5/md5.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/network/util.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/nuki_lock/nuki_lock.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
In file included from /data/esp32-nuki/.piolibdeps/esp32-nuki/Nuki Ble for Esp32/src/NukiBle.h:9,
                 from src/esphome/components/nuki_lock/nuki_lock.h:9,
                 from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
/data/esp32-nuki/.piolibdeps/esp32-nuki/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/restart/button/restart_button.cpp.o
Compiling /data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/sensor/automation.cpp.o
*** [/data/esp32-nuki/.pioenvs/esp32-nuki/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================= [FAILED] Took 36.83 seconds =========================

```

any ideas?

axute, 2022-12-14:
It works with the dev branch, great.
And the door opens too ;-)
Thanks so much!
Do you need the new logfile?

uriyacovy, 2022-12-14:
Not yet, but thanks.

ruimarinho, 2022-12-16:
@uriyacovy do you think with the release of 2022.12.0 the Nuki Lock could be supported directly via Active Bluetooth Proxies? Exchanged a message with @bdraco and he mentioned:

> it looks like HA already has all the frameworks needed to talk to these locks so anyone should be able to make an integration for them without any additional changes to HA core

uriyacovy, 2022-12-17:
As far as I understand, Active Bluetooth Proxies are managed by SW running on HA. This is different then what is implemented here (the edge SW on the esp running the logic and bluetooth communication).
You might want to check with the following projects if they would implement your idea: [RaspiNukiBridge](https://github.com/regevbr/RaspiNukiBridge), [NukiBridgeAddon](https://github.com/ftarolli/NukiBridgeAddon)

mundschenk-at, 2022-12-17:
Not sure how the bridge implementations would implement a HA integration that interacts with the lock via its BT API? (I understand you see this as out of scope for this project, but for the bridge projects it will be even more so.)

uriyacovy, 2022-12-17:
The bridge SW implements the interaction with the NUKI lock over standard BT libraries. It only needs to add support for esphome BT proxy.

mundschenk-at, 2022-12-17:
Yeah but I think its unlikely that they want to add a dependency on HA/ESPhome when their interface is the "home automation-neutral" Nuki Bridge API.

mundschenk-at, 2022-12-17:
(I mean the add-on might in principle be willing, but I doubt there is a way for an add-on running as a separate docker container to easily access the proxied BT API.)

ruimarinho, 2023-01-01:
As far as I understand the Home Assistant BT integration can now manage active BT connections as well, all via bluetooth proxies using ESPHome. My idea is that I can use any ESP32 I have available, flash with a generic ESPHome build with active connection proxying to Home Assistant, and add Nuki on Home Assistant directly.

Very similar to what is being done with SwitchBot devices -- https://github.com/home-assistant/core/blob/dev/homeassistant/components/switchbot/__init__.py.

The advantage is that there would be no specific ESPHome build to replace the Nuki Bridge, only code to support it directly in Home Assistant.

uriyacovy, 2023-01-02:
@ruimarinho, as I wrote before, I agree that this a valid approach, but this module does not implement the bride SW, only the ESPHome wrapper.

uriyacovy, 2023-01-29:
Should be fixed at the main branch now.

closed 2023-01-29 by uriyacovy