uriyacovy/ESPHome_nuki_lock #16: requestConfig and requestKeyTurnerState failed

LucaDiba, 2022-12-28:
Hi! I'm using the dev version of this integration and I'm experiencing some availability issues.
Sometimes the Nuki goes offline and I'm not able to access any sensor data nor lock/unlock the Nuki.

These are the logs:
```
03:15:39 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:15:49 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:15:49 | [D] | [api:102] | Accepted ::FFFF:10.0.0.200
03:16:00 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:16:10 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:16:10 | [W] | [api.connection:081] | Home Assistant 2022.12.8 (::FFFF:10.0.0.200): Connection reset
03:16:20 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:16:30 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:16:40 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:16:50 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:16:50 | [D] | [api:102] | Accepted ::FFFF:10.0.0.200
03:17:00 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:17:10 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:17:10 | [D] | [api.connection:918] | Home Assistant 2022.12.8 (::FFFF:10.0.0.200): Connected successfully
03:17:20 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:17:30 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:17:30 | [W] | [api.connection:081] | Home Assistant 2022.12.8 (::FFFF:10.0.0.200): Connection reset
03:17:40 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
03:17:50 | [E] | [nukilock.lock:087] | print_keypad_entries: requestConfig failed (result 3)
03:18:00 | [E] | [nukilock.lock:075] | requestKeyTurnerState failed: 3
```

This is my yaml:
```
esphome:
  name: mw-th-0-floor
  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32#dev

esp32:
  board: mhetesp32devkit
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:
api:
web_server:
  port: 80

lock:
  - platform: nuki_lock
    name: Porta soggiorno
    is_connected:
      name: "Porta soggiorno is connected"
    is_paired: 
      name: "Porta soggiorno is paired"
    battery_level:
      name: "Porta soggiorno batteria"
```

These are some screenshots of the ESP web page.

Nothing:
[image]

Accepted:
[image]

Connected successfully:
[image]

Connection reset:
[image]

uriyacovy, 2022-12-29:
What's the reason you call print_keypad_entries so frequently?

LucaDiba, 2022-12-29:
I don't call it at all. This all happens without me doing anything.

uriyacovy, 2022-12-29:
OK, you are right and the printouts are from a wrong string in the code.
Back to the original problem - this happens on BT communication errors or when the ESP crashes.
If the ESP is close enough to the NUKI, please disable the web server and check the availability.

zanna-37, 2023-01-05:
I have the same issue. If I send 2 commands (open or close) near each other the ESP32 no longer works and I need to manually unplug it.
This doesn't happen with the main branch. In fact, in the branch main I receive the error but the ESP32 continues working.

uriyacovy, 2023-02-11:
@zanna-37, is the issue resolved now?

zanna-37, 2023-02-11:
Nope, I solved the infinite loop with [what I stated in the other issue](https://github.com/uriyacovy/ESPHome_nuki_lock/issues/18#issuecomment-1409977106). However the problem persists. It has less impact, but still it is impossibile to communicate with the lock for some seconds after a quick succession of 2 or more commands.

Honestly, now that I read the code, I think that this issue should be taken care of in the library that you use. But I don't known the project enough to be sure.

uriyacovy, 2023-02-11:
It should be quite easy to run the NukiBLE sample and check if this reproduced there as well.

harmptor, 2023-05-30:
Is there any news on this? I'm currently using the main branch and my ESP still crashes when sending "open" commands too quickly (5 seconds). Should I be using the dev branch?

uriyacovy, 2023-06-02:
Nothing new yet. You can follow up also here https://github.com/I-Connect/NukiBleEsp32/issues/37
How often do you send 2 open command within 5 seconds?

harmptor, 2023-06-02:
Thank you, I'm now following that thread. I send one open command every 5 seconds until my door sensor detects open (it's a reed contact sensor, attached to the ESP board running the nuki firmware)

closed 2024-10-29 by uriyacovy