uriyacovy/ESPHome_nuki_lock #17: Compilation issue under ESPHome - home assistant 2022.12.8

Jejebond, 2022-12-31:
Hi, i've got this issue when compiling under ESPHome - home assistant 2022.12.8

`Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/nuki_lock/nuki_lock.cpp.o
Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/socket/bsd_sockets_impl.cpp.o
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
src/esphome/components/nuki_lock/nuki_lock.h:50:50: error: 'Nuki::LockState' has not been declared
         lock::LockState nuki_to_lock_state(Nuki::LockState);
                                                  ^~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.h:65:9: error: 'BleScanner' does not name a type
         BleScanner scanner_;
         ^~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.h:66:15: error: 'KeyTurnerState' in namespace 'Nuki' does not name a type
         Nuki::KeyTurnerState retrievedKeyTurnerState_;
               ^~~~~~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:7:52: error: 'esphome::lock::LockState esphome::nuki_lock::NukiLock::nuki_to_lock_state' is not a static data member of 'class esphome::nuki_lock::NukiLock'
 lock::LockState NukiLock::nuki_to_lock_state(Nuki::LockState nukiLockState) {
                                                    ^~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:7:52: error: 'LockState' is not a member of 'Nuki'
src/esphome/components/nuki_lock/nuki_lock.cpp:7:52: note: suggested alternative: 'CommandState'
 lock::LockState NukiLock::nuki_to_lock_state(Nuki::LockState nukiLockState) {
                                                    ^~~~~~~~~
                                                    CommandState
src/esphome/components/nuki_lock/nuki_lock.cpp: In member function 'void esphome::nuki_lock::NukiLock::update_status()':
src/esphome/components/nuki_lock/nuki_lock.cpp:55:38: error: 'class Nuki::NukiBle' has no member named 'requestKeyTurnerState'
     uint8_t result = this->nukiBle_->requestKeyTurnerState(&(this->retrievedKeyTurnerState_));
                                      ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:55:68: error: 'class esphome::nuki_lock::NukiLock' has no member named 'retrievedKeyTurnerState_'
     uint8_t result = this->nukiBle_->requestKeyTurnerState(&(this->retrievedKeyTurnerState_));
                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:1:
src/esphome/components/nuki_lock/nuki_lock.cpp:58:17: error: 'class esphome::nuki_lock::NukiLock' has no member named 'retrievedKeyTurnerState_'
           this->retrievedKeyTurnerState_.criticalBatteryState,
                 ^~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:105:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:57:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Bat state: %#x, Bat crit: %d, Bat perc:%d lock state: %d %d:%d:%d",
         ^~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:59:27: error: 'class Nuki::NukiBle' has no member named 'isBatteryCritical'
           this->nukiBle_->isBatteryCritical(), this->nukiBle_->getBatteryPerc(), this->retrievedKeyTurnerState_.lockState,
                           ^~~~~~~~~~~~~~~~~
src/esphome/core/log.h:105:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:57:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Bat state: %#x, Bat crit: %d, Bat perc:%d lock state: %d %d:%d:%d",
         ^~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:59:64: error: 'class Nuki::NukiBle' has no member named 'getBatteryPerc'
           this->nukiBle_->isBatteryCritical(), this->nukiBle_->getBatteryPerc(), this->retrievedKeyTurnerState_.lockState,
                                                                ^~~~~~~~~~~~~~
src/esphome/core/log.h:105:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:57:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Bat state: %#x, Bat crit: %d, Bat perc:%d lock state: %d %d:%d:%d",
         ^~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:59:88: error: 'class esphome::nuki_lock::NukiLock' has no member named 'retrievedKeyTurnerState_'
           this->nukiBle_->isBatteryCritical(), this->nukiBle_->getBatteryPerc(), this->retrievedKeyTurnerState_.lockState,
                                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:105:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^~~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:57:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Bat state: %#x, Bat crit: %d, Bat perc:%d lock state: %d %d:%d:%d",
         ^~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:60:17: error: 'class esphome::nuki_lock::NukiLock' has no member named 'retrievedKeyTurnerState_'
           this->retrievedKeyTurnerState_.currentTimeHour,
                 ^~~~~~~~~~~~~~~~~~~~~~~~
[... 62 lines cut ...]
           ^~~~~~~~
           handler_
src/esphome/components/nuki_lock/nuki_lock.cpp:120:48: error: cannot convert 'Nuki::PairingResult' to 'bool' in initialization
         bool paired = this->nukiBle_->pairNuki();
                                                ^
src/esphome/components/nuki_lock/nuki_lock.cpp: In member function 'virtual void esphome::nuki_lock::NukiLock::control(const esphome::lock::LockCall&)':
src/esphome/components/nuki_lock/nuki_lock.cpp:136:38: error: 'class Nuki::NukiBle' has no member named 'lockAction'; did you mean 'executeAction'?
             result = this->nukiBle_->lockAction(Nuki::LockAction::Lock);
                                      ^~~~~~~~~~
                                      executeAction
src/esphome/components/nuki_lock/nuki_lock.cpp:136:55: error: 'Nuki::LockAction' has not been declared
             result = this->nukiBle_->lockAction(Nuki::LockAction::Lock);
                                                       ^~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:138:38: error: 'class Nuki::NukiBle' has no member named 'lockAction'; did you mean 'executeAction'?
             result = this->nukiBle_->lockAction(this->open_latch_ ? Nuki::LockAction::Unlatch : Nuki::LockAction::Unlock);
                                      ^~~~~~~~~~
                                      executeAction
src/esphome/components/nuki_lock/nuki_lock.cpp:138:75: error: 'Nuki::LockAction' has not been declared
             result = this->nukiBle_->lockAction(this->open_latch_ ? Nuki::LockAction::Unlatch : Nuki::LockAction::Unlock);
                                                                           ^~~~~~~~~~
src/esphome/components/nuki_lock/nuki_lock.cpp:138:103: error: 'Nuki::LockAction' has not been declared
             result = this->nukiBle_->lockAction(this->open_latch_ ? Nuki::LockAction::Unlatch : Nuki::LockAction::Unlock);
                                                                                                       ^~~~~~~~~~
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================= [FAILED] Took 15.38 seconds =========================`

uriyacovy, 2023-01-02:
You need to use the dev branch for now. See here:
https://github.com/uriyacovy/ESPHome_nuki_lock/issues/15#issuecomment-1351419332

uriyacovy, 2023-01-29:
Can you try again with the main branch?

uriyacovy, 2023-04-27:
Should be solved.

closed 2023-04-27 by uriyacovy