uriyacovy/ESPHome_nuki_lock #18: Crash under ESPhome 2022.12.5

lichtl, 2023-01-23:
Hi, just upgraded for latest 12.5 due to some BLE fixes and it seems that those changes might have broken the current implementation. Unlocking/Locking the door is no longer working and the ESP seems to reset. State reporting seems to work fine.

```
INFO Starting log output from flur-nuki.local using esphome API
INFO Successfully connected to flur-nuki.local
[10:45:31][I][app:102]: ESPHome version 2022.12.5 compiled on Jan 23 2023, 10:41:19
[10:45:31][C][wifi:504]: WiFi:
[10:45:31][C][wifi:362]:   Local MAC: EC:62:60:93:35:A8
[10:45:31][C][wifi:363]:   SSID: 'FRITZ!Box 7530 JP'[redacted]
[10:45:31][C][wifi:364]:   IP Address: 192.168.178.157
[10:45:31][C][wifi:366]:   BSSID: 44:4E:6D:6E:DB:5D[redacted]
[10:45:31][C][wifi:367]:   Hostname: 'flur-nuki'
[10:45:31][C][wifi:369]:   Signal strength: -71 dB ▂▄▆█
[10:45:31][C][wifi:373]:   Channel: 1
[10:45:31][C][wifi:374]:   Subnet: 255.255.255.0
[10:45:31][C][wifi:375]:   Gateway: 192.168.178.1
[10:45:31][C][wifi:376]:   DNS1: 192.168.178.1
[10:45:31][C][wifi:377]:   DNS2: 0.0.0.0
[10:45:31][C][logger:293]: Logger:
[10:45:31][C][logger:294]:   Level: DEBUG
[10:45:31][C][logger:295]:   Log Baud Rate: 115200
[10:45:31][C][logger:296]:   Hardware UART: UART0
[10:45:31][C][nukilock.lock:330]: nukilock.lockNuki Lock 'Nuki Lock'
[10:45:31][C][nukilock.lock:331]: nukilock.lockIs Connected 'Nuki Connected'
[10:45:31][C][nukilock.lock:331]: nukilock.lock  Device Class: 'connectivity'
[10:45:31][C][nukilock.lock:332]: nukilock.lockIs Paired 'Nuki Paired'
[10:45:31][C][nukilock.lock:332]: nukilock.lock  Device Class: 'connectivity'
[10:45:31][C][nukilock.lock:333]: nukilock.lockBattery Critical 'Nuki Battery Critical'
[10:45:31][C][nukilock.lock:333]: nukilock.lock  Device Class: 'battery'
[10:45:31][C][nukilock.lock:334]: nukilock.lockDoor Sensor 'Nuki Door Sensor'
[10:45:31][C][nukilock.lock:334]: nukilock.lock  Device Class: 'door'
[10:45:31][C][nukilock.lock:335]: nukilock.lockDoor Sensor State 'Nuki Door Sensor State'
[10:45:31][C][nukilock.lock:336]: nukilock.lockBattery Level 'Nuki Battery Level'
[10:45:31][C][nukilock.lock:336]: nukilock.lock  Device Class: 'battery'
[10:45:31][C][nukilock.lock:336]: nukilock.lock  State Class: ''
[10:45:31][C][nukilock.lock:336]: nukilock.lock  Unit of Measurement: '%'
[10:45:31][C][nukilock.lock:336]: nukilock.lock  Accuracy Decimals: 0
[10:45:31][C][nukilock.lock:337]: Unpair request is false
[10:45:31][C][captive_portal:088]: Captive Portal:
[10:45:31][C][mdns:103]: mDNS:
[10:45:31][C][mdns:104]:   Hostname: flur-nuki
[10:45:31][C][ota:093]: Over-The-Air Updates:
[10:45:31][C][ota:094]:   Address: flur-nuki.local:3232
[10:45:31][C][ota:097]:   Using Password.
[10:45:31][C][api:138]: API Server:
[10:45:31][C][api:139]:   Address: flur-nuki.local:6053
[10:45:31][C][api:141]:   Using noise encryption: YES
[10:46:00][I][nukilock.lock:025]: event notified 0
[10:46:01][I][nukilock.lock:025]: event notified 0
[10:46:01][I][nukilock.lock:025]: event notified 0
[10:46:01][I][nukilock.lock:025]: event notified 0
[10:46:01][I][nukilock.lock:025]: event notified 0
[10:46:01][I][nukilock.lock:025]: event notified 0
[10:46:01][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
[10:46:01][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[10:46:01][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[10:46:02][E][nukilock.lock:087]: print_keypad_entries: requestConfig failed (result 2)
[10:46:02][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
[10:46:03][I][nukilock.lock:063]: Bat state: 0xa4, Bat crit: 0, Bat perc:82 lock state: 1 9:46:3
[10:46:03][D][lock:055]: 'Nuki Lock': Sending state LOCKED
[10:46:03][D][binary_sensor:036]: 'Nuki Connected': Sending state ON
[10:46:03][D][sensor:127]: 'Nuki Battery Level': Sending state 82.00000 % with 0 decimals of accuracy
[10:46:03][D][text_sensor:067]: 'Nuki Door Sensor State': Sending state 'unavailable'
[10:46:18][D][api:102]: Accepted ::FFFF:192.168.178.11
[10:46:18][D][api.connection:918]: Home Assistant 2023.1.7 (::FFFF:192.168.178.11): Connected successfully
[10:46:29][E][nukilock.lock:194]: lockAction failed: 2
[10:46:29][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[10:46:29][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[10:46:31][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
[10:46:32][E][nukilock.lock:087]: print_keypad_entries: requestConfig failed (result 2)
[10:46:35][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
INFO flur-nuki.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO flur-nuki.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for flur-nuki.local
WARNING Disconnected from API
INFO Successfully connected to flur-nuki.local
```

lichtl, 2023-01-26:
Did some more investigation and after a repairing it at least work once (when sending a unlock the esphome seems to crash) see below logs

```
INFO Reading configuration /config/esphome/flur-nuki.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Starting log output from flur-nuki.local using esphome API
INFO Successfully connected to flur-nuki.local
[12:10:14][I][app:102]: ESPHome version 2022.12.7 compiled on Jan 26 2023, 12:05:39
[12:10:14][C][wifi:504]: WiFi:
[12:10:14][C][wifi:362]:   Local MAC: EC:62:60:93:35:A8
[12:10:14][C][wifi:363]:   SSID: 'FRITZ!Box 7530 JP'[redacted]
[12:10:14][C][wifi:364]:   IP Address: 192.168.178.157
[12:10:14][C][wifi:366]:   BSSID: 44:4E:6D:6E:DB:5D[redacted]
[12:10:14][C][wifi:367]:   Hostname: 'flur-nuki'
[12:10:14][C][wifi:369]:   Signal strength: -74 dB ▂▄▆█
[12:10:14][C][wifi:373]:   Channel: 1
[12:10:14][C][wifi:374]:   Subnet: 255.255.255.0
[12:10:14][C][wifi:375]:   Gateway: 192.168.178.1
[12:10:14][C][wifi:376]:   DNS1: 192.168.178.1
[12:10:14][C][wifi:377]:   DNS2: 0.0.0.0
[12:10:14][C][logger:293]: Logger:
[12:10:14][C][logger:294]:   Level: DEBUG
[12:10:14][C][logger:295]:   Log Baud Rate: 115200
[12:10:14][C][logger:296]:   Hardware UART: UART0
[12:10:14][C][nukilock.lock:330]: nukilock.lockNuki Lock 'Nuki Lock'
[12:10:14][C][nukilock.lock:331]: nukilock.lockIs Connected 'Nuki Connected'
[12:10:14][C][nukilock.lock:331]: nukilock.lock  Device Class: 'connectivity'
[12:10:14][C][nukilock.lock:332]: nukilock.lockIs Paired 'Nuki Paired'
[12:10:14][C][nukilock.lock:332]: nukilock.lock  Device Class: 'connectivity'
[12:10:14][C][nukilock.lock:333]: nukilock.lockBattery Critical 'Nuki Battery Critical'
[12:10:14][C][nukilock.lock:333]: nukilock.lock  Device Class: 'battery'
[12:10:14][C][nukilock.lock:334]: nukilock.lockDoor Sensor 'Nuki Door Sensor'
[12:10:14][C][nukilock.lock:334]: nukilock.lock  Device Class: 'door'
[12:10:14][C][nukilock.lock:335]: nukilock.lockDoor Sensor State 'Nuki Door Sensor State'
[12:10:14][C][nukilock.lock:336]: nukilock.lockBattery Level 'Nuki Battery Level'
[12:10:14][C][nukilock.lock:336]: nukilock.lock  Device Class: 'battery'
[12:10:14][C][nukilock.lock:336]: nukilock.lock  State Class: ''
[12:10:14][C][nukilock.lock:336]: nukilock.lock  Unit of Measurement: '%'
[12:10:14][C][nukilock.lock:336]: nukilock.lock  Accuracy Decimals: 0
[12:10:14][C][nukilock.lock:337]: Unpair request is false
[12:10:14][C][captive_portal:088]: Captive Portal:
[12:10:14][C][mdns:103]: mDNS:
[12:10:14][C][mdns:104]:   Hostname: flur-nuki
[12:10:14][C][ota:093]: Over-The-Air Updates:
[12:10:14][C][ota:094]:   Address: flur-nuki.local:3232
[12:10:14][C][ota:097]:   Using Password.
[12:10:14][C][api:138]: API Server:
[12:10:14][C][api:139]:   Address: flur-nuki.local:6053
[12:10:14][C][api:141]:   Using noise encryption: YES
[12:10:23][D][api:102]: Accepted ::FFFF:192.168.178.11
[12:10:23][D][api.connection:918]: Home Assistant 2023.1.7 (::FFFF:192.168.178.11): Connected successfully
[12:10:31][D][lock:055]: 'Nuki Lock': Sending state LOCKED
[12:10:48][I][nukilock.lock:025]: event notified 0
[12:10:49][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
[12:10:49][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[12:10:49][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[12:10:51][E][nukilock.lock:087]: print_keypad_entries: requestConfig failed (result 2)
[12:10:53][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
[12:10:55][I][nukilock.lock:063]: Bat state: 0xa4, Bat crit: 0, Bat perc:82 lock state: 1 11:10:59
[12:10:55][D][lock:055]: 'Nuki Lock': Sending state LOCKED
[12:10:55][D][binary_sensor:036]: 'Nuki Connected': Sending state ON
[12:10:55][D][sensor:127]: 'Nuki Battery Level': Sending state 82.00000 % with 0 decimals of accuracy
[12:10:55][D][text_sensor:067]: 'Nuki Door Sensor State': Sending state 'unavailable'
[12:12:37][D][lock:055]: 'Nuki Lock': Sending state UNLOCKED
[12:13:03][E][nukilock.lock:194]: lockAction failed: 3
[12:13:03][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[12:13:03][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
INFO flur-nuki.local: Ping timed out!
INFO Disconnected from ESPHome API for flur-nuki.local
WARNING Disconnected from API
```

uriyacovy, 2023-01-29:
Can you try again with the main branch?

zanna-37, 2023-01-30:
For me it goes in a loop forever when issuing 2 commands near to each other.

```
[22:33:23][V][api.connection:900]: Hello from client: 'Home Assistant 2023.1.7 (HA_IP)' | API Version 1.7
[22:33:23][D][api.connection:918]: Home Assistant 2023.1.7 (HA_IP): Connected successfully
[22:33:32][D][lock:055]: 'Nuki Stairs lock': Sending state UNLOCKED
[22:33:32][V][component:200]: Component api took a long time for an operation (1.45 s).
[22:33:32][V][component:201]: Components should block for at most 20-30ms.
[22:33:47][E][nukilock.lock:180]: lockAction failed: 3
[22:33:47][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[22:33:47][D][lock:055]: 'Nuki Stairs lock': Sending state UNKNOWN
[22:33:47][V][component:200]: Component api took a long time for an operation (10.08 s).
[22:33:47][V][component:201]: Components should block for at most 20-30ms.
[22:33:57][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[22:33:57][V][component:200]: Component nuki_lock.lock took a long time for an operation (10.10 s).
[22:33:57][V][component:201]: Components should block for at most 20-30ms.
[22:33:57][D][api:102]: Accepted HA_IP
[22:34:07][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[22:34:07][V][component:200]: Component nuki_lock.lock took a long time for an operation (10.10 s).
[22:34:07][V][component:201]: Components should block for at most 20-30ms.
[22:34:07][W][api.connection:081]: Home Assistant 2023.1.7 (HA_IP): Connection reset
[22:34:17][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[22:34:17][V][component:200]: Component nuki_lock.lock took a long time for an operation (10.07 s).
[22:34:17][V][component:201]: Components should block for at most 20-30ms.
[22:34:17][V][api:114]: Removing connection to Home Assistant 2023.1.7 (HA_IP)
[22:34:17][V][api.connection:900]: Hello from client: 'Home Assistant 2023.1.7 (HA_IP)' | API Version 1.7
[22:34:27][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[22:34:27][V][component:200]: Component nuki_lock.lock took a long time for an operation (10.09 s).
[22:34:27][V][component:201]: Components should block for at most 20-30ms.
[22:34:27][D][api:102]: Accepted HA_IP
[22:34:27][W][api.connection:081]: Home Assistant 2023.1.7 (HA_IP): Connection reset
[22:34:38][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[22:34:38][V][component:200]: Component nuki_lock.lock took a long time for an operation (10.10 s).
[22:34:38][V][component:201]: Components should block for at most 20-30ms.
[22:34:38][V][api:114]: Removing connection to Home Assistant 2023.1.7 (HA_IP)
[22:34:48][E][nukilock.lock:075]: requestKeyTurnerState failed: 3
[22:34:48][V][component:200]: Component nuki_lock.lock took a long time for an operation (10.08 s).
[22:34:48][V][component:201]: Components should block for at most 20-30ms.
```

uriyacovy, 2023-01-31:
Hi @zanna-37,
Looks like the Nuki is not connected to the ESP (the function calls timed out). Try to unpair and pair again and see if this works.

zanna-37, 2023-01-31:
That's not the case because the first command always works. And if I reboot the ESP32, it works again.

By the way, I forked the project and fix it myself. I'm using an updated version of `https://github.com/I-Connect/NukiBleEsp32` (which, at the time of writing, is at `0ccb839`) because yours is now 114 (!) commits behind from the upstream repo.
Then I removed the following 2 lines from `components/nuki_lock/nuki_lock.cpp`
```
    this->nukiLock_->setConnectTimeout(BLE_CONNECT_TIMEOUT_SEC);
    this->nukiLock_->setConnectRetries(BLE_CONNECT_TIMEOUT_RETRIES);
```
Everything is working since. I can only test the lock, no other Nuki accessories.

uriyacovy, 2023-02-11:
@zanna-37, I'm trying to get my PR into the NukiBleEsp32 then we should be able to use the updated repo.
Closing this issue for now.

closed 2023-02-11 by uriyacovy

wes1993, 2023-05-07:
Hello @zanna-37,
ha you found a solution for this problem?
I have tried your fork but i still have this error... here is my log:
```
[11:58:04][D][lock:042]: 'Nuki Lock' Opening.
[11:58:34][I][nukilock.lock:025]: event notified 0
[11:58:39][E][nukilock.lock:075]: requestKeyTurnerState failed: 2
[11:58:39][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[11:58:39][D][lock:054]: 'Nuki Lock': Sending state UNKNOWN
[11:58:44][I][nukilock.lock:063]: Bat state: 0xc8, Bat crit: 0, Bat perc:100 lock state: 3 9:58:44
[11:58:44][D][lock:054]: 'Nuki Lock': Sending state UNLOCKED
[11:58:44][D][binary_sensor:036]: 'Nuki Connected': Sending state ON
[11:58:44][D][sensor:110]: 'Nuki Battery Level': Sending state 100.00000 % with 0 decimals of accuracy
[11:58:44][D][text_sensor:064]: 'Nuki Door Sensor State': Sending state 'unavailable'
```

Do you have some ideas?
Also i don't understand why if in the string we have the lock state, in my log is state 3 but the Nuki Door Sensor State is unavailable..

Best Regards
Stefano

zanna-37, 2023-05-13:
I don't have a nuki door sensor, so I can't help you, I'm sorry. The repo works fine for the lock in my tests.