uriyacovy/ESPHome_nuki_lock #20: NukiLock.h missing

hsafra, 2023-01-29:
Hi @uriyacovy and thanks for your hard work here.

I'm trying to install the new version out of the main branch and get an error saying 

```
Compiling /data/door-fingerprint/.pioenvs/door-fingerprint/src/esphome/components/nuki_lock/nuki_lock.cpp.o
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
src/esphome/components/nuki_lock/nuki_lock.h:10:10: fatal error: NukiLock.h: No such file or directory

```
Full log

```
INFO Reading configuration /config/esphome/door-fingerprint.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing door-fingerprint (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- Preferences @ 2.0.0
|-- ESP32 BLE Arduino @ 1.0.1+sha.adc2aee
|-- Nuki Ble for Esp32 @ 0.0.4+sha.b0f8588
|   |-- NimBLE-Arduino @ 1.4.1+sha.0d9f039
|   |-- CRC16 @ 0.0.0+20230129215632.sha.f46d86b
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/door-fingerprint/.pioenvs/door-fingerprint/src/esphome/components/nuki_lock/nuki_lock.cpp.o
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
src/esphome/components/nuki_lock/nuki_lock.h:10:10: fatal error: NukiLock.h: No such file or directory

******************************************************************
* Looking for NukiLock.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NukiLock.h"
* Web  > https://registry.platformio.org/search?q=header:NukiLock.h
*
******************************************************************

 #include "NukiLock.h"
          ^~~~~~~~~~~~
compilation terminated.
*** [/data/door-fingerprint/.pioenvs/door-fingerprint/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================== [FAILED] Took 3.00 seconds ==========================
```

I have an older version installed on the ESP and waited for the main branch's update to update it.

Thanks

hsafra, 2023-01-30:
I tried to use the #dev branch like so:
```
esphome:
  name: door-fingerprint
  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32#dev

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock
```
And it compiled successfully.

Could this be an issue with the main branch?

uriyacovy, 2023-01-30:
Did you update the yaml when moving to the new main version? It should be identical to the dev branch.

hsafra, 2023-01-30:
yaml file attached below. 
I have a fingerprint sensor attached to the ESP too, secrets are redacted

```
esphome:
  name: door-fingerprint
  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXX"
  services:
  - service: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - service: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: delete_all
    then:
      - fingerprint_grow.delete_all:

ota:
  password: "XXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Door-Fingerprint"
    password: "XXXXX"

captive_portal:

uart:
  tx_pin: GPIO16
  rx_pin: GPIO05
  baud_rate: 57600
    
    
#http_request:
#  useragent: esphome/device
#  timeout: 5s
  
sensor:
  - platform: fingerprint_grow
    fingerprint_count:
      name: "Fingerprint Count"
    last_finger_id:
      name: "Fingerprint Last Finger ID"
    last_confidence:
      name: "Fingerprint Last Confidence"
    status:
      name: "Fingerprint Status"
    capacity:
      name: "Fingerprint Capacity"
    security_level:
      name: "Fingerprint Security Level"
  - platform: uptime
    name: ESPHome door-fingerprint Uptime Sensor
      
text_sensor:
  - platform: template
    id: fingerprint_state
    name: "Fingerprint State"
    
fingerprint_grow:
  password: XXXXX
  sensing_pin: GPIO19
  on_finger_scan_matched:
    - logger.log: "Authorized finger, unlocking"
    - fingerprint_grow.aura_led_control:
        state: ALWAYS_ON
        speed: 0
        color: CYAN
        count: 0
    - script.execute: unlock_front_door
    - delay: 0.5s
    - if:
        condition:
          lock.is_locked: front_door
        then:
          - logger.log: "Unlocking second time"
          - script.execute: unlock_front_door
    - delay: 0.5s
    - if:
        condition:
          lock.is_locked: front_door
        then:
          - logger.log: "Unlocking third time"
          - script.execute: unlock_front_door
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Authorized finger"
  on_finger_scan_unmatched:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 2
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Unauthorized finger"
  on_enrollment_scan:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: BLUE
        count: 2
    - fingerprint_grow.aura_led_control:
        state: ALWAYS_ON
        speed: 0
        color: PURPLE
        count: 0
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Finger scanned"
  on_enrollment_done:
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 100
        color: BLUE
        count: 2
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Enrolled fingerprint"
  on_enrollment_failed:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 4
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Failed to enroll fingerprint"
        

script:
  - id: unlock_front_door
    mode: restart     # Start a new run after first stopping previous run.
    then:
      - lock.unlock: front_door
      - fingerprint_grow.aura_led_control:
          state: BREATHING
          speed: 200
          color: BLUE
          count: 1
        
lock:
  - platform: nuki_lock
    name: Front door
    id: front_door
    is_paired: 
      name: "Nuki Paired"
    is_connected: 
      name: "Nuki Connected"
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
      state_class: "measurement"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"
```
I tried to add
```
    version: 2.0.3
    platform_version: 4.4.0
```
like the example but no change

Thanks,
Harel

hsafra, 2023-01-30:
Thanks @uriyacovy, that was indeed the issue.

I actually removed the #dev suffix on purpose since it was under the "How to use -- Dev Branch --" header on the main page and I thought it was only appropriate to the dev branch and not the main one.
[image]

Thanks for your help

closed 2023-01-30 by hsafra

uriyacovy, 2023-01-30:
Yes, indeed, this was an error in the title, and it is fixed now.