uriyacovy/ESPHome_nuki_lock #21: 'NimBLEClient' in namespace 'esphome::ble_client' does not name a type

cirolaferrara, 2023-02-09:
Yaml:
```
esphome:
  name: airthings-ble
  libraries:
    - Preferences
    - https://github.com/uriyacovy/NukiBleEsp32#dev

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Airthings-Ble Fallback Hotspot"
    password: ""

captive_portal:

logger:
  #level: DEBUG
  level: VERBOSE

switch:
  - platform: restart
    name: "Airthings-ble Restart"
    
sensor:
  # Xiaomi
  - platform: xiaomi_miscale
    mac_address: 'XX:XX:XX:XX:XX:XX'
    weight:
      name: "Xiaomi Mi Scale"
      
  # Airthings Wave Plus
  - platform: airthings_wave_plus
    ble_client_id: airthings
    update_interval: 5min # default
    temperature:
      name: "WavePlus Temperature"
    radon:
      name: "WavePlus Radon"
    radon_long_term:
      name: "WavePlus Radon Long Term"
    pressure:
      name: "WavePlus Pressure"
    humidity:
      name: "WavePlus Humidity"
    co2:
      name: "WavePlus CO2"
    tvoc:
      name: "WavePlus VOC"

ble_client:
  # Airthings Wave Plus
  - mac_address: 'XX:XX:XX:XX:XX:XX'
    id: airthings

esp32_ble_tracker:
  scan_parameters:
    interval: 1.2s
    window: 500ms

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"
```

Log:

> Compiling /data/airthings-ble/.pioenvs/airthings-ble/libdbe/NimBLE-Arduino/NimBLEAddress.cpp.o
> In file included from /data/airthings-ble/.piolibdeps/airthings-ble/NukiBleEsp/src/NukiBle.h:14,
>                  from /data/airthings-ble/.piolibdeps/airthings-ble/NukiBleEsp/src/NukiLock.h:3,
>                  from src/esphome/components/nuki_lock/nuki_lock.h:10,
>                  from src/esphome.h:40,
>                  from src/main.cpp:3:
> Compiling /data/airthings-ble/.pioenvs/airthings-ble/libdbe/NimBLE-Arduino/NimBLEAdvertisedDevice.cpp.o
> /data/airthings-ble/.piolibdeps/airthings-ble/NimBLE-Arduino/src/NimBLEDevice.h:54:41: error: 'NimBLEClient' in namespace 'esphome::ble_client' does not name a type
>  #define BLEClient                       NimBLEClient
>                                          ^~~~~~~~~~~~
> src/main.cpp:27:13: note: in expansion of macro 'BLEClient'
>  ble_client::BLEClient *airthings;
>              ^~~~~~~~~
> src/main.cpp:27:1: note: suggested alternative: 'BLEClient'
>  ble_client::BLEClient *airthings;
>  ^~~~~~~~~~
>  BLEClient
> src/main.cpp: In function 'void setup()':
> src/main.cpp:322:3: error: 'airthings' was not declared in this scope
>    airthings = new ble_client::BLEClient();
>    ^~~~~~~~~
> src/main.cpp:322:19: error: expected type-specifier
>    airthings = new ble_client::BLEClient();
>                    ^~~~~~~~~~
> Compiling /data/airthings-ble/.pioenvs/airthings-ble/libdbe/NimBLE-Arduino/NimBLEAdvertising.cpp.o
> *** [/data/airthings-ble/.pioenvs/airthings-ble/src/main.cpp.o] Error 1
> ========================= [FAILED] Took 197.92 seconds =========================

uriyacovy, 2023-02-11:
@cirolaferrara, Nuki lock is using a different BLE (Bluetooth) library, and I guess it conflicts with the standard BLEClient of ESPHome.
Try to run the lock only and see if it works.

cirolaferrara, 2023-02-12:
Lock only works but I need airthings_wave_plus.

uriyacovy, 2023-02-12:
Unfortunately, adding additional BT devices is not support (at least at the moment).

closed 2023-02-12 by uriyacovy