uriyacovy/ESPHome_nuki_lock #23: Build fails,

ScIT-Raphael, 2023-03-21:
I currently try to build esphome nuki lock on HASS, esphome v2023.3.0 but get the following error:

```
...
Compiling /data/node-nuki/.pioenvs/node-nuki/libdec/NukiBleEsp/NukiUtils.cpp.o
/data/node-nuki/.piolibdeps/node-nuki/NukiBleEsp/src/NukiUtils.cpp:4:10: fatal error: Crc16.h: No such file or directory

***************************************************************
* Looking for Crc16.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Crc16.h"
* Web  > https://registry.platformio.org/search?q=header:Crc16.h
*
***************************************************************

 #include "Crc16.h"
          ^~~~~~~~~
compilation terminated.
*** [/data/node-nuki/.pioenvs/node-nuki/libdec/NukiBleEsp/NukiUtils.cpp.o] Error 1
========================= [FAILED] Took 29.64 seconds =========================
```

Does anyone have an idea, how to solve or where I could have a look?

Zelbi, 2023-03-21:
Hi, Same error, didn't find solution yet.

uriyacovy, 2023-03-22:
Can you please state the HA and ESPHome version you are using?

ScIT-Raphael, 2023-03-22:
Used esphome yaml:

```
esphome:
  name: node-nuki
  friendly_name: node-nuki
  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32#dev

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Node-Nuki Fallback Hotspot"
    password: "xxx"

captive_portal:
  
lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"
```

niahane, 2023-03-22:
Hi, i am ssh'ed into home assistant and esphome docker image i have copied this 2 libraries in the source files of NukiBleEsp now compilation works fine.
root@15ef4d2f-esphome:/data/nuki-test/.piolibdeps/nuki-test/NukiBleEsp/src# cp ../../CRC16/src/Crc16.cpp ./
root@15ef4d2f-esphome:/data/nuki-test/.piolibdeps/nuki-test/NukiBleEsp/src# cp ../../CRC16/src/Crc16.h ./

For me is the first time using it, i have pressed the nuki door v3 button until white ring appear 

I have added esphome entity to home assistant but i have only a entry about firmware update and nothing else. I try many times the pairing but nothing happen, the white ring after several seconds turn off and no new entities appear in esphome integration sensor. Esphome log dont say nothing relevant.

niahane, 2023-03-22:
I have do my tests with 2 diffrent esp32 board, an olimex-poe and a esp32-wroom devkit c4.

Esphome wireless log not reporting nothing interensting.

I have connected my esp32 board to pc and start log from usb tty, now i have a decent log:

[ 36791][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[ 37287][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[ 37789][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[ 38291][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[ 38787][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[ 39289][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[ 39791][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND

IN THIS MOMENT I HAVE TRIGGERED NUKI PAIRING BUTTON:

connectBle retry 0
pairNuki: connectBle success after 986 ms
sendPlainMessage: connectBle success after 0 ms
E (54104) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (54104) task_wdt:  - loopTask (CPU 1)
E (54104) task_wdt: Tasks currently running:
E (54104) task_wdt: CPU 0: IDLE
E (54104) task_wdt: CPU 1: IDLE
E (54104) task_wdt: Aborting.

abort() was called at PC 0x400fe824 on core 0

Backtrace:0x40083709:0x3ffbea0c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:258]: Log initialized
[C][ota:469]: There have been 6 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][nukilock.lock:084]: Starting NUKI Lock...
[   348][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[W][nukilock.lock:109]: ���? Nuki is not paired
[D][binary_sensor:034]: 'Nuki Paired': Sending initial state OFF
[D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: 7C:9E:BD:30:19:74
[D][wifi:386]: Starting scan...
[   416][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[   720][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  1220][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  1720][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  2220][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  2720][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[  3220][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND

When i trigger pairing my Esp32 boards crash :'(

I had to open a different issue for this?

uriyacovy, 2023-03-23:
ESPHome bumped the platform_version to 5.3.0 (see https://github.com/esphome/esphome/pull/4254). 
Please change the following and build again (if it doesn't work, try also to delete '.esphome' folder and build again):
Change platform_version to: `platform_version: 5.3.0`
In libraries just after Preferences add a line with 
`- https://github.com/vinmenn/Crc16.git`

Please let me know if this works for you.

ScIT-Raphael, 2023-03-23:
No luck, adjusted like suggested, removed .esphome aswell, still the same error:

```
esphome:
  name: node-nuki
  friendly_name: node-nuki
  libraries:
  - Preferences
  - https://github.com/uriyacovy/NukiBleEsp32#dev
  - https://github.com/vinmenn/Crc16.git

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.3.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx="

ota:
  password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Node-Nuki Fallback Hotspot"
    password: "xxxx"

captive_portal:
  
lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"
```

uriyacovy, 2023-03-23:
The crc16 lib should appear just after Preferences:
```
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32#dev

```

ScIT-Raphael, 2023-03-23:
You're right, wrong order. Build worked now, unlucky I get the following error in a endless loop:

`[ 12418][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND`

niahane, 2023-03-23:
OK! I followed your instructions and now everything works well. Thank you! :-)

uriyacovy, 2023-03-23:
@ScIT-Raphael, is the Nuki paired? can you post the full log?

ScIT-Raphael, 2023-03-23:
Somehow pairing didnt worked out on first try, now all is working. Awesome, thank you!

closed 2023-03-23 by uriyacovy