uriyacovy/ESPHome_nuki_lock #25: Not a issue. I ask about the door opening.

niahane, 2023-03-23:
Hi, my door lock has 3 modes, locked, unlocked and door opening.  There is a way to send the door opening signal to the nuki, as can be done from the app? In home assistant the lock entity has only 2 options, lock, unlock and there is a service that unlock for some time and relock... In the official app there is a mode called "Open the door" if i use it, the nuki turn itself at the maximum of anticlockwise rotation and my door open like someone use door handle.

Thank You.

uriyacovy, 2023-03-23:
Yes, this is done with open service. See here: https://github.com/uriyacovy/ESPHome_nuki_lock/issues/1

niahane, 2023-03-23:
Wow it work perfectly, very very thanks! This integration is awesome.