uriyacovy/ESPHome_nuki_lock #26: On lock, ESPHome Nuki Lock loses connection and shows lockAction failed: 99

ScIT-Raphael, 2023-03-24:
As soon as I want to lock the nuki over home assistant, "Nuki Connected" drops to "Disconnected" and the logs show the following errors:

```
[07:56:34][E][nukilock.lock:180]: lockAction failed: 99
[07:56:34][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[07:56:34][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[07:56:35][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[07:56:35][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[07:56:36][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[07:56:36][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[07:56:37][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
```

Do you maybe have an idea how to solve?

uriyacovy, 2023-03-24:
Most chances are that the esp's BT is too far from the lock. Did you move the esp after pairing?

ScIT-Raphael, 2023-03-24:
No, location is the same - around 2.5 meters away from nuki, just pulled it to 1 meter distance - no change.

uriyacovy, 2023-03-24:
Please post the full log from startup and I'll have a look.

niahane, 2023-03-27:
Same here, [08:33:14][E][nukilock.lock:180]: lockAction failed: 99
[08:33:14][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[08:33:14][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[08:33:14][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:15][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:15][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:16][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:16][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:17][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:17][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:18][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:18][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:19][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:19][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:20][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:20][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:21][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:21][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:22][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:22][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:33:22][I][nukilock.lock:025]: event notified 0
[08:33:24][I][nukilock.lock:063]: Bat state: 0xc8, Bat crit: 0, Bat perc:100 lock state: 3 6:33:26

Often it works but sometimes it gives me this error too.

uriyacovy, 2023-03-28:
Can you test the sample from the base library [NukiBleEsp32](https://github.com/I-Connect/NukiBleEsp32)?

ScIT-Raphael, 2023-03-28:
Sorry for the delay, have changed to the suggested library but havent got any change. Here is the full log:

```
INFO Reading configuration /config/esphome/node-nuki.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Starting log output from node-nuki.local using esphome API
INFO Successfully connected to node-nuki.local
[08:21:51][I][app:102]: ESPHome version 2023.3.2 compiled on Mar 28 2023, 08:19:41
[08:21:51][C][wifi:504]: WiFi:
[08:21:51][C][wifi:362]:   Local MAC: [redacted]
[08:21:51][C][wifi:363]:   SSID: [redacted]
[08:21:51][C][wifi:364]:   IP Address: [redacted]
[08:21:51][C][wifi:366]:   BSSID: [redacted]
[08:21:51][C][wifi:367]:   Hostname: 'node-nuki'
[08:21:51][C][wifi:369]:   Signal strength: -69 dB ▂▄▆█
[08:21:51][C][wifi:373]:   Channel: 6
[08:21:51][C][wifi:374]:   Subnet: 255.255.254.0
[08:21:51][C][wifi:375]:   Gateway: [redacted]
[08:21:51][C][wifi:376]:   DNS1: [redacted]
[08:21:51][C][wifi:377]:   DNS2: 0.0.0.0
[08:21:51][C][logger:293]: Logger:
[08:21:51][C][logger:294]:   Level: DEBUG
[08:21:51][C][logger:295]:   Log Baud Rate: 115200
[08:21:51][C][logger:296]:   Hardware UART: UART0
[08:21:51][C][nukilock.lock:193]: nukilock.lockNuki Lock 'Nuki Lock'
[08:21:51][C][nukilock.lock:194]: nukilock.lockIs Connected 'Nuki Connected'
[08:21:51][C][nukilock.lock:194]: nukilock.lock  Device Class: 'connectivity'
[08:21:51][C][nukilock.lock:195]: nukilock.lockIs Paired 'Nuki Paired'
[08:21:51][C][nukilock.lock:195]: nukilock.lock  Device Class: 'connectivity'
[08:21:51][C][nukilock.lock:196]: nukilock.lockBattery Critical 'Nuki Battery Critical'
[08:21:51][C][nukilock.lock:196]: nukilock.lock  Device Class: 'battery'
[08:21:51][C][nukilock.lock:197]: nukilock.lockDoor Sensor 'Nuki Door Sensor'
[08:21:51][C][nukilock.lock:197]: nukilock.lock  Device Class: 'door'
[08:21:51][C][nukilock.lock:198]: nukilock.lockDoor Sensor State 'Nuki Door Sensor State'
[08:21:51][C][nukilock.lock:199]: nukilock.lockBattery Level 'Nuki Battery Level'
[08:21:51][C][nukilock.lock:199]: nukilock.lock  Device Class: 'battery'
[08:21:51][C][nukilock.lock:199]: nukilock.lock  State Class: ''
[08:21:51][C][nukilock.lock:199]: nukilock.lock  Unit of Measurement: '%'
[08:21:51][C][nukilock.lock:199]: nukilock.lock  Accuracy Decimals: 0
[08:21:51][C][captive_portal:088]: Captive Portal:
[08:21:53][C][mdns:108]: mDNS:
[08:21:53][C][mdns:109]:   Hostname: node-nuki
[08:23:16][E][nukilock.lock:180]: lockAction failed: 99
[08:23:16][D][binary_sensor:036]: 'Nuki Connected': Sending state OFF
[08:23:16][D][lock:055]: 'Nuki Lock': Sending state UNKNOWN
[08:23:16][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:17][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:17][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:18][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:18][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:19][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:19][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:20][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:20][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:21][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:21][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[08:23:22][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
... looping
```

uriyacovy, 2023-03-29:
@ScIT-Raphael, do you mean that it does not work with [NukiBleEsp32](https://github.com/I-Connect/NukiBleEsp32) sample?
If so, I suggest to open the issue there as well. This component is built on top of NukiBleEsp32.

stefanschaedeli, 2023-05-06:
I have exactly the same behaviour. I did even repair sucessfully, it did work again for a few hours and then I have the error again. Strang thing is that it still reports connected...
[logs_esp-display_logs.txt](https://github.com/uriyacovy/ESPHome_nuki_lock/files/11412459/logs_esp-display_logs.txt)

flosch-dev, 2023-05-08:
after some time, 1-2 hours getting again the same problem when trying to lock the nuki.

```
[10:33:35][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[10:33:35][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[10:33:36][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[10:33:36][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[10:33:37][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[10:33:37][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
[10:33:38][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
```

icherfas, 2023-08-23:
Hi,
I'm having the same issue.
it's happening right from the start after I pair the devices.
few more things I have noticed:
1. if I change the state of the nuki (from the lock itself for example) it will not update the status in ha.
2. I have tried to different esp32 and I get the same results.
3. if I reset the esp32 then the status of the lock is correct (but will not update if it changes)
4. as soon as I try to change status from the ha, the esp disconnect from the nuki and I get the 99 error

thanks.

philipgo, 2023-09-10:
My device worked flawlessly for several months, all of a sudden it stopped working. The only interesting part in the logs is
'''
[00:46:01][W][component:204]: Component nuki_lock.lock took a long time for an operation (0.05 s).
[00:46:01][W][component:205]: Components should block for at most 20-30ms.
[00:46:01][E][nukilock.lock:075]: requestKeyTurnerState failed: 99
'''
Things I tried without success:
- New ESP32 hardware
- fresh batteries in the Nuki Lock
- Move ESP32 within 1m of the lock (used to work flawlessly over a distance of ca 10m)
- switched to the dev branch

lienbacher, 2023-10-09:
I have the same problem and additionally a problem with `[E][nukilock.lock:191]: lockAction failed: 2` and `[E][nukilock.lock:191]: lockAction failed: 3` errors. 

Here's a full log of me just trying to open and close the lock. Reliability is below 33%.
<details>
  <summary>Full log:</summary> 

 ```[12:40:16][I][app:102]: ESPHome version 2023.9.3 compiled on Oct  9 2023, 12:39:36
[12:40:16][C][wifi:546]: WiFi:
[12:40:16][C][wifi:382]:   Local MAC: A8:42:E3:CA:58:A4
[12:40:16][C][wifi:383]:   SSID: [redacted]
[12:40:16][C][wifi:384]:   IP Address: 192.168.0.77
[12:40:16][C][wifi:386]:   BSSID: [redacted]
[12:40:16][C][wifi:387]:   Hostname: 'nuki-haustuer'
[12:40:16][C][wifi:389]:   Signal strength: -57 dB ▂▄▆█
[12:40:16][C][wifi:393]:   Channel: 1
[12:40:16][C][wifi:394]:   Subnet: 255.255.255.0
[12:40:16][C][wifi:395]:   Gateway: 192.168.0.1
[12:40:16][C][wifi:396]:   DNS1: 192.168.0.1
[12:40:16][C][wifi:397]:   DNS2: 0.0.0.0
[12:40:16][C][logger:357]: Logger:
[12:40:16][C][logger:358]:   Level: DEBUG
[12:40:16][C][logger:359]:   Log Baud Rate: 115200
[12:40:16][C][logger:361]:   Hardware UART: UART0
[12:40:16][C][nukilock.lock:327]: nukilock.lockNuki Lock 'Schloss'
[12:40:16][C][nukilock.lock:328]: nukilock.lockIs Connected 'Connected'
[12:40:16][C][nukilock.lock:328]: nukilock.lock  Device Class: 'connectivity'
[12:40:16][C][nukilock.lock:329]: nukilock.lockIs Paired 'Paired'
[12:40:16][C][nukilock.lock:329]: nukilock.lock  Device Class: 'connectivity'
[12:40:16][C][nukilock.lock:330]: nukilock.lockBattery Critical 'Battery Critical'
[12:40:16][C][nukilock.lock:330]: nukilock.lock  Device Class: 'battery'
[12:40:16][C][nukilock.lock:333]: nukilock.lockBattery Level 'Battery Level'
[12:40:16][C][nukilock.lock:333]: nukilock.lock  Device Class: 'battery'
[12:40:16][C][nukilock.lock:333]: nukilock.lock  State Class: ''
[12:40:16][C][nukilock.lock:333]: nukilock.lock  Unit of Measurement: '%'
[12:40:16][C][nukilock.lock:333]: nukilock.lock  Accuracy Decimals: 0
[12:40:16][C][nukilock.lock:334]: Unpair request is false
[12:40:16][C][version.text_sensor:021]: Version Text Sensor 'ESPHome Version'
[12:40:16][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[12:40:17][C][captive_portal:088]: Captive Portal:
[12:40:17][C][mdns:115]: mDNS:
[12:40:17][C][mdns:116]:   Hostname: nuki-haustuer
[12:40:17][C][ota:097]: Over-The-Air Updates:
[12:40:17][C][ota:098]:   Address: nuki-haustuer.local:3232
[12:40:17][C][ota:101]:   Using Password.
[12:40:17][C][api:138]: API Server:
[12:40:17][C][api:139]:   Address: nuki-haustuer.local:6053
[12:40:17][C][api:141]:   Using noise encryption: YES
[12:40:17][C][wifi_signal.sensor:009]: WiFi Signal 'Signal'
[12:40:17][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[12:40:17][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[12:40:17][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[12:40:17][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[12:40:17][C][wifi_info:009]: WifiInfo IPAddress 'IP'
[12:40:17][C][wifi_info:011]: WifiInfo SSID 'SSID'
[12:40:18][D][api:102]: Accepted 192.168.0.101
[12:40:18][D][api.connection:1031]: ESPHome Logs 2023.9.3 (192.168.0.101): Connected successfully
[12:40:40][E][nukilock.lock:191]: lockAction failed: 2
[12:40:40][D][binary_sensor:036]: 'Connected': Sending state OFF
[12:40:40][D][lock:054]: 'Schloss': Sending state UNKNOWN
[12:40:40][W][component:204]: Component api took a long time for an operation (3.05 s).
[12:40:40][W][component:205]: Components should block for at most 20-30ms.
[12:40:41][I][nukilock.lock:063]: Bat state: 0xac, Bat crit: 0, Bat perc:86 lock state: 3 10:40:41
[12:40:41][D][lock:054]: 'Schloss': Sending state UNLOCKED
[12:40:41][D][binary_sensor:036]: 'Connected': Sending state ON
[12:40:41][D][sensor:094]: 'Battery Level': Sending state 86.00000 % with 0 decimals of accuracy
[12:40:41][W][component:204]: Component nuki_lock.lock took a long time for an operation (1.27 s).
[12:40:41][W][component:205]: Components should block for at most 20-30ms.
[12:41:14][D][sensor:094]: 'Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[12:42:14][D][sensor:094]: 'Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[12:43:14][D][sensor:094]: 'Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[... 188 lines cut ...]
[12:52:54][E][nukilock.lock:191]: lockAction failed: 2
[12:52:54][D][binary_sensor:036]: 'Connected': Sending state OFF
[12:52:54][D][lock:054]: 'Schloss': Sending state UNKNOWN
[12:52:54][W][component:204]: Component api took a long time for an operation (3.06 s).
[12:52:54][W][component:205]: Components should block for at most 20-30ms.
[12:52:56][I][nukilock.lock:063]: Bat state: 0xac, Bat crit: 0, Bat perc:86 lock state: 3 10:52:55
[12:52:56][D][lock:054]: 'Schloss': Sending state UNLOCKED
[12:52:56][D][binary_sensor:036]: 'Connected': Sending state ON
[12:52:56][D][sensor:094]: 'Battery Level': Sending state 86.00000 % with 0 decimals of accuracy
[12:52:56][W][component:204]: Component nuki_lock.lock took a long time for an operation (1.90 s).
[12:52:56][W][component:205]: Components should block for at most 20-30ms.
[12:52:56][D][sensor:094]: 'Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[12:53:09][D][lock:054]: 'Schloss': Sending state LOCKED
[12:53:10][W][component:204]: Component api took a long time for an operation (1.96 s).
[12:53:10][W][component:205]: Components should block for at most 20-30ms.
[... cut ...]
[12:53:26][I][nukilock.lock:063]: Bat state: 0xac, Bat crit: 0, Bat perc:86 lock state: 1 10:53:26
[12:53:26][D][sensor:094]: 'Battery Level': Sending state 86.00000 % with 0 decimals of accuracy
[12:53:26][W][component:204]: Component nuki_lock.lock took a long time for an operation (1.73 s).
[12:53:26][W][component:205]: Components should block for at most 20-30ms.
[12:53:55][D][sensor:094]: 'Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[12:54:55][D][sensor:094]: 'Signal': Sending state -57.00000 dBm with 0 decimals of accuracy
 ```
  
</details>

lienbacher, 2023-10-09:
So I actually ditched this component and went for the [nuki_hub](https://github.com/technyon/nuki_hub) as in this current state your esphome component is production ready for me. No problems with nuki_hub at all so far. 

From what I know about esphome the problem might very much be related to these lines: 
```
[12:52:54][W][component:204]: Component api took a long time for an operation (3.06 s).
[12:52:54][W][component:205]: Components should block for at most 20-30ms.
```

mrbluebrett, 2023-12-12:
Same problem here...

uriyacovy, 2023-12-13:
Hi,
Though I tried to find some time to support this module, I'm unable to invest the required attention. Hopefully some community member will take the lead. Sorry.

closed 2024-10-29 by uriyacovy