uriyacovy/ESPHome_nuki_lock #28: Build failed - esp_err.h: No such file or directory

flosch-dev, 2023-04-20:
Hello, just got new Nuki 3 and wanted to use this nice implementation with ESPHome.
Unfortuantely the build with ESP Home within Home Assistant failed.
I tried to use ESPHome dev branch and also removing the arduino version as. 
I hope this is not a stupid problem on my side, I use other ESPhome devices already, but never with external libraries. Compiling the others work fine.

I'm running following version:
```
Home Assistant 2023.4.5
Supervisor 2023.04.0
Operating System 10.0
Frontend 20230411.1 - latest
```

ESPHome:
```
Current version: 2023.4.0
```
My ESPhome yaml looks like this:

```
esphome:
  name: nuki
  friendly_name: Nuki
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.3.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: XXX
  password: XXX

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nuki Fallback Hotspot"
    password: "XXX"

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"

captive_portal:
    
```

When compiling, I get following errors:
```
INFO Reading configuration /config/esphome/nuki.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing nuki (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/data/cache/platformio/packages/framework-arduinoespressif32@3.20003.220626/tools/platformio-build-esp32.py'
File "/data/cache/platformio/packages/framework-arduinoespressif32@3.20003.220626/tools/platformio-build.py", line 92, in <module>
Library Manager: Installing FS
Warning! Could not find the package with 'FS' requirements for your system 'linux_x86_64'
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_x86_64'
Library Manager: Installing ESPmDNS
Warning! Could not find the package with 'ESPmDNS' requirements for your system 'linux_x86_64'
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- Preferences @ 2.1.0
|-- Crc16 @ 0.1.2+sha.719a7a1
|-- NukiBleEsp @ 2.0.0+sha.1de6290
|   |-- NimBLE-Arduino @ 1.4.1
|   |-- BleScanner @ 1.1.0+sha.575385f
|   |   |-- NimBLE-Arduino @ 1.4.1
|-- WiFi @ 1.2.7
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/api/api_connection.cpp.o
Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/nuki/.pioenvs/nuki/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:127:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/log.h:14:10: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:127:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/api/api_pb2.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/log.h:14:10: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================== [FAILED] Took 2.56 seconds ==========================
```

uriyacovy, 2023-04-27:
I just tried now with HA 2023.4.6 and ESPHome 2023.4.2 and it builds without an issue. Can you try deleting your ```.esphome``` folder and build again?

flosch-dev, 2023-04-27:
Thanks for trying at yours!
Since there have been some updates of esphome these days, I was always trying to compile again, but no luck. I moved the folder .esphome, but it does not help :(
I also recreate new esphome device, but always getting the same issue. 

```INFO Reading configuration /config/esphome/nuki.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing nuki (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing git+https://github.com/vinmenn/Crc16.git
INFO Installing git+https://github.com/vinmenn/Crc16.git
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-c74_2wlz'...
Library Manager: Crc16@0.1.2+sha.719a7a1 has been installed!
INFO Crc16@0.1.2+sha.719a7a1 has been installed!
Library Manager: Installing git+https://github.com/uriyacovy/NukiBleEsp32
INFO Installing git+https://github.com/uriyacovy/NukiBleEsp32
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-3b5sisq9'...
Library Manager: NukiBleEsp@2.0.0+sha.1de6290 has been installed!
INFO NukiBleEsp@2.0.0+sha.1de6290 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing h2zero/NimBLE-Arduino @ ^1.4.0
INFO Installing h2zero/NimBLE-Arduino @ ^1.4.0
Unpacking  [####################################]  100%
Library Manager: NimBLE-Arduino@1.4.1 has been installed!
INFO NimBLE-Arduino@1.4.1 has been installed!
Library Manager: Installing git+https://github.com/vinmenn/Crc16
INFO Installing git+https://github.com/vinmenn/Crc16
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-x8zsm8m6'...
Library Manager: CRC16@0.1.2+sha.719a7a1 has been installed!
INFO CRC16@0.1.2+sha.719a7a1 has been installed!
Library Manager: Installing git+https://github.com/I-Connect/BleScanner
INFO Installing git+https://github.com/I-Connect/BleScanner
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-slb173xd'...
Library Manager: BleScanner@1.1.0+sha.575385f has been installed!
INFO BleScanner@1.1.0+sha.575385f has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
[... 129 lines cut ...]
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/api/api_pb2.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/log.h:14:10: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
*** [/data/nuki/.pioenvs/nuki/src/esphome/components/api/api_pb2_service.cpp.o] Error 1```

uriyacovy, 2023-04-28:
The compilation error doesn't look related to any of the things I'm using. We're you able to build other esphome projects? What yaml did you use?

flosch-dev, 2023-05-02:
Just another esphome update today, but still build failed. Yes, I can build all the other esphome devices. The difference to the others is that I never include external components, not sure if it is related to that.

The YAML looks like this:

```
esphome:
  name: nuki
  friendly_name: Nuki
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.3.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "..."

captive_portal:

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"```

uriyacovy, 2023-05-03:
@flosch-dev, on which operating system do you build the project?

flosch-dev, 2023-05-03:
I'm using Home Assistant Operating System on x86-64 platform.

uriyacovy, 2023-05-06:
I apologize, but I don't have this setup and can't reproduce the issue. Maybe someone else could.

flosch-dev, 2023-05-07:
so far I was using the add-on of ESPHome within HomeAssistant. However as it seems I'm the only one with that problem, I tried now to build with ESPHome Docker image (official) and indeed the build was successful.
Not sure what the problem with the HomeAssistant is.
I will keep posted if I find out what was wrong there.

Thanks