uriyacovy/ESPHome_nuki_lock #34: Failed to compile with esp-idf

reydecopas, 2023-08-31:
working with arduino framework but when trying to upgrade to esp-idf framework for another board (shelly 1 plus) fails to compile

Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ntc/ntc.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/nuki_lock/nuki_lock.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_arduino_esp32.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_arduino_esp8266.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_arduino_rp2040.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_esp_idf.o
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
src/esphome/components/nuki_lock/nuki_lock.h:10:10: fatal error: NukiLock.h: No such file or directory

******************************************************************
* Looking for NukiLock.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NukiLock.h"
* Web  > https://registry.platformio.org/search?q=header:NukiLock.h
*
******************************************************************

 #include "NukiLock.h"
          ^~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/shelly1plus-nuki/src/esphome/components/nuki_lock/nuki_lock.o] Error 1
========================== [FAILED] Took 7.03 seconds ==========================

zipposi, 2023-09-01:
I have the same problem.

Latest HASS: 2023.8.x
Latest EspHome. 2023.8.2

zipposi, 2023-09-01:
Solved!

Changed the line

`  - https://github.com/I-Connect/NukiBleEsp32#0.0.4`

to

`  - https://github.com/I-Connect/NukiBleEsp32`

zipposi, 2023-09-01:
I was too quick :(

https://github.com/uriyacovy/ESPHome_nuki_lock/issues/35

reydecopas, 2023-09-02:
No luck

config.yaml
[...]
esphome:
  name: shelly1plus-nuki
  platformio_options:
    board_build.f_cpu: 160000000L
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/I-Connect/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"
[...]

INFO ESPHome 2023.8.2
INFO Reading configuration /config/shelly1plus-nuki.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing shelly1plus-nuki (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/vinmenn/Crc16.git
INFO Installing git+https://github.com/vinmenn/Crc16.git
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-asqr0tq5'...
Library Manager: Crc16@0.1.2+sha.719a7a1 has been installed!
INFO Crc16@0.1.2+sha.719a7a1 has been installed!
Library Manager: Installing git+https://github.com/I-Connect/NukiBleEsp32
INFO Installing git+https://github.com/I-Connect/NukiBleEsp32
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-ihdtwvfg'...
Library Manager: NukiBleEsp@2.0.0+sha.707b327 has been installed!
INFO NukiBleEsp@2.0.0+sha.707b327 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing h2zero/NimBLE-Arduino @ ^1.4.0
INFO Installing h2zero/NimBLE-Arduino @ ^1.4.0
Unpacking  [####################################]  100%
Library Manager: NimBLE-Arduino@1.4.1 has been installed!
INFO NimBLE-Arduino@1.4.1 has been installed!
Library Manager: Installing git+https://github.com/vinmenn/Crc16
INFO Installing git+https://github.com/vinmenn/Crc16
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-27c_yr3x'...
Library Manager: CRC16@0.1.2+sha.719a7a1 has been installed!
INFO CRC16@0.1.2+sha.719a7a1 has been installed!
Library Manager: Installing git+https://github.com/I-Connect/BleScanner
INFO Installing git+https://github.com/I-Connect/BleScanner
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-n_t6h53l'...
Library Manager: BleScanner@1.1.0+sha.2b6c045 has been installed!
INFO BleScanner@1.1.0+sha.2b6c045 has been installed!
Library Manager: Resolving dependencies...
[... 64 lines cut ...]
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/nuki_lock/nuki_lock.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_arduino_esp32.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_arduino_esp8266.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_arduino_rp2040.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_backend_esp_idf.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/ota/ota_component.o
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/resistance/resistance_sensor.o
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
src/esphome/components/nuki_lock/nuki_lock.h:10:10: fatal error: NukiLock.h: No such file or directory

******************************************************************
* Looking for NukiLock.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NukiLock.h"
* Web  > https://registry.platformio.org/search?q=header:NukiLock.h
*
******************************************************************

 #include "NukiLock.h"
          ^~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/restart/switch/restart_switch.o
*** [.pioenvs/shelly1plus-nuki/src/esphome/components/nuki_lock/nuki_lock.o] Error 1
Compiling .pioenvs/shelly1plus-nuki/src/esphome/components/sensor/automation.o
========================= [FAILED] Took 18.55 seconds =========================

jubaraspy, 2023-09-27:
I have the same problem...

closed 2024-08-08 by uriyacovy