uriyacovy/ESPHome_nuki_lock #40: Crash on Pairing

AzonInc, 2024-03-25:
Whenever I'm trying to pair the smart lock the ESP is crashing and goes into a bootloop.
I have the Smartlock 4.0 Pro with Wifi disabled.

Might be related to #24 

Hardware: NodeMCU ESP32

Config:
```
substitutions:
  name: tcs-intercom-test
  friendly_name: TCS Intercom Test

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

web_server:

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@dev
  
esp32:
  board: "nodemcu-32s"  # Or whatever other board you're using
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.3.0

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"

  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"

  #  unpair: True
```

Logs:
```
[ 45805][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
[... 276 lines cut ...]
E (13717) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13717) task_wdt:  - loopTask (CPU 1)
E (13717) task_wdt: Tasks currently running:
E (13717) task_wdt: CPU 0: IDLE
E (13717) task_wdt: CPU 1: IDLE
E (13717) task_wdt: Aborting.

abort() was called at PC 0x40102964 on core 0

Backtrace:0x40083709:0x3ffbea0c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
```

JakobTewes, 2024-04-02:
Heyhey @AzonInc,

as I don´t even get it to pairing...maybe you can help me ;-). How did you pair the NUKI?
I tried it with "press-button-on-Nuki-until-ring-lights-up" and reboot the ESP32.
The NUKI had a pin set in the beginning, then i reset it and did not set a pin (but set it up via app).

Thanks and regards

Jakob

AzonInc, 2024-04-02:
> Heyhey @AzonInc,
>
> as I don´t even get it to pairing...maybe you can help me ;-). How did you pair the NUKI? I tried it with "press-button-on-Nuki-until-ring-lights-up" and reboot the ESP32. The NUKI [...]

Actually I tried it several times until it "paired" and failed :(
I did it like u said - just pressing the button on the nuki for a few seconds until it lights up and then it immediately tried to pair and crashed right afterwards into a bootloop.

I'm using my fork of the project with a few additions to be able to initiate pairing mode myself so it doesn't try to pair all the time and only when neccessary.
However it doesnt make any difference while using the dev branch here also fails.

Maybe it's somewhat related to the 4th generation Locks.

Did you try nuki_hub already? That worked for me, even tho the connection was terrible.

JakobTewes, 2024-04-02:
Ah, nice to read, I’m not going wrong ;-)
Can you tell, when pairing in the original (dev branch) code happens? is it timed or on boot?

Got a gen1 lock so maybe we can proof the theory regarding the lock generation?

AzonInc, 2024-04-03:
The pairing happens whenever possible starting after initial component setup.
Your's doesn't crash right?

closed 2024-04-04 by AzonInc