uriyacovy/ESPHome_nuki_lock #41: Pairing unclear

JakobTewes, 2024-03-30:
Heyhey @uriyacovy or collaborateur, 

I got uncertainties regarding pairing the Nuki lock to this. Can someone plz explain,
as I still got no luck after resetting the Nuki, pressing the button on the lock like an idiot 
and rebooting the esp multiple times ;-)

Can someone plz help?

Thanks and regards

AzonInc, 2024-04-04:
The pairing happens whenever possible starting after initial component setup.
You could give the `pair-and-unpair-refactor` branch of NukiBleEsp32 a shot.

Just change the esphome libraries block:
```
esphome:
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32#pair-and-unpair-refactor
```

JakobTewes, 2024-04-14:
Thanks @AzonInc. Changed to the pair-and-unpair-refactor branch.
Unfortunately still got no success. Can you advice on how to pair?
   At what configuration status of the lock can I pair?
   Do i need to reset the lock?
   Can I add it to the Nuki App (while not pairing to any bridge)?
   Is there a thing to keep in mind (like setting no pin via App)?

Thanks again and regards

Jakob

AzonInc, 2024-04-14:
> At what configuration status of the lock can I pair?

I never tried in any other state than after it's already paired with the nuki app.

> Can I add it to the Nuki App (while not pairing to any bridge)?

Isn't it already?

> Is there a thing to keep in mind (like setting no pin via App)?

I have a Pin set, however I don't know if it affects pairing.
I added the PIN Functionality in my Fork because I thought this could be the issue but seems to be working fine without as well.

> Are there known incompatibilities with other BLE applications (esp32_ble_beacon, bluetooth_proxy, esp32_ble_tracker)

I think it's not possible to use them concurrently, at least ESPHome doesn't allow me to compile it.

JakobTewes, 2024-04-19:
Uh....sorry to insist, but already pressed my fingers through the lock.
Still not getting it to pair....when does it happen? 

- I press the button until it lights up (NUKI V1)
- Wait and nothing happens. No logs

Alos tried:
- I press the button until it lights up (NUKI V1)
- Rebooting the ESP
- Wait and nothing happens. No logs

uriyacovy, 2024-10-26:
@JakobTewes please try with the latest version and reopen the issue if this still doesn't work.

closed 2024-10-26 by uriyacovy