uriyacovy/ESPHome_nuki_lock #42: Activity log

szisso, 2024-04-17:
Is there support of activity log?
I would like to get the name of the device that lock/unlock the door.
While searching the web I saw that authorizationName is the value that use by the Nukihub components but I couldn’t find documentation on the esphome version

AzonInc, 2024-10-26:
Are you still looking for log support?
Would be nice to have someone testing the new feature.

szisso, 2024-10-26:
Yes I am still need itShai Zisso‫ב-26 באוק׳

‬

AzonInc, 2024-10-26:
Okay so lets give the auth-logs branch a shot.

```yaml
esphome:
  libraries:
  - Preferences
  - https://github.com/h2zero/NimBLE-Arduino#1.4.0
  - Crc16
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://AzonInc/ESPHome_nuki_lock@feature/auth-logs
```

AzonInc, 2024-10-29:
I guess this can be closed as the changes have been merged :)

closed 2024-10-29 by uriyacovy