uriyacovy/ESPHome_nuki_lock #44: Error after last Esphome version 2024.5.0

ploto22, 2024-05-16:
After last esphome version 2024.5.0, there are the follow compile error about library register:

`
Compiling .pioenvs/air-door/src/esphome/components/nuki_lock/nuki_lock.cpp.o
In file included from .piolibdeps/air-door/NukiBleEsp/src/NukiLock.h:3,
                 from src/esphome/components/nuki_lock/nuki_lock.h:10,
                 from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
.piolibdeps/air-door/NukiBleEsp/src/NukiBle.h:14:10: fatal error: NimBLEDevice.h: No such file or directory

**********************************************************************
* Looking for NimBLEDevice.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NimBLEDevice.h"
* Web  > https://registry.platformio.org/search?q=header:NimBLEDevice.h
*
**********************************************************************

 #include "NimBLEDevice.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/air-door/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================= [FAILED] Took 59.81 seconds =========================
========== [ERROR] /config/esphome/air-door.yaml ==========
````

How can resolve the error ?
Thanks

marioedani, 2024-05-17:
I have the same problem. Already here #43

marioedani, 2024-05-17:
I confirm that reverting ESPHome back to previous version (2024.4.2) the problem is gone

uriyacovy, 2024-05-18:
I suspect that this is because ESPHome changed platformio [version](https://github.com/esphome/esphome/pull/6634). Please check if adding `h2zero/NimBLE-Arduino` to libraries section solves the problem:
```
  libraries:
  - Preferences
  - https://github.com/h2zero/NimBLE-Arduino
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32
```

marioedani, 2024-05-18:
> I suspect that this is because ESPHome changed platformio [version](https://github.com/esphome/esphome/pull/6634). Please check if adding `h2zero/NimBLE-Arduino` to libraries section solves the [...]

Tested now.
Adding "NimBLE-Arduino" repository to libraries list solves the issue! 👍 
Thank you very much for your support

ploto22, 2024-05-18:
Great issue resolved !  - CLOSED -
Be attention: use the same sequence of the libraries, if reversed again the issue 
Thank you very much for your support

the-mentor, 2024-06-01:
This issue can be closed since it was fixed in PR https://github.com/uriyacovy/ESPHome_nuki_lock/pull/48

uriyacovy, 2024-06-04:
Solved in rel v0.0.3

closed 2024-06-04 by uriyacovy