uriyacovy/ESPHome_nuki_lock #49: Support for Smart Lock v4

axute, 2024-06-03:
I tried last week to pair with Smart Lock v4, but it doesn't work.
Are there any plans to support this?
Nukihub works with v4, but I would like to use Esphome.

pazzotranquillo, 2024-06-14:
Same request here! Thanks.

ssorgatem, 2024-06-24:
Another v4 owner here

uriyacovy, 2024-06-27:
Apologies for the late response. AFAIK, the underlying library [NukiBleEsp32](https://github.com/I-Connect/NukiBleEsp32) does not yet support v4.

ssorgatem, 2024-06-28:
Its author says it does: https://github.com/I-Connect/NukiBleEsp32/issues/65

uriyacovy, 2024-06-28:
Great. There is no specific code in this module for either V2, V3 and V4.
Since I don't have the V4 version, can you explain what is the problem you experience?

axute, 2024-06-28:
It is simply not pairing.

uriyacovy, 2024-06-29:
Please post the logs and the yaml.

axute, 2024-07-04:
Unfortunately I can't test it anymore, my dad is happy with the nukihub project, sorry.
I only have a Nuki V3 myself.

ssorgatem, 2024-07-09:
I will be able to test once https://github.com/uriyacovy/ESPHome_nuki_lock/issues/51 is fixed

uriyacovy, 2024-07-26:
> I will be able to test once #51 is fixed

You can try now.

AntorFr, 2024-08-12:
Adding more verbose logs

[17:24:29]I[37400][D][NukiBle.cpp:109] pairNuki(): Nuki in pairing mode found
[17:24:29] NimBLEScan: Updated advertiser: b8:7c:6f:7b:91:e3
[17:24:29]D NimBLEScan: >> stop()
[17:24:29]I NimBLEAdvertisedDevice: No service data found
[17:24:29]D NimBLEScan: << stop()
[17:24:29]I N[ 37419][D][NukiBle.cpp:163] connectBle(): connecting within: loopTask
[17:24:29][ 37426][D][NukiBle.cpp:177] connectBle(): connection attemnpt 0
[17:24:29]imBLEScan: Updated advertiser: e7:a7:29:e1:39:c8
[17:24:29]D NimBLEClient: >> connect(54:d2:72:93:48:bd)
[17:24:29]D NimBLEClient: Got Client event 
[17:24:29]I NimBLEClient: Connected event
[17:24:29]D NimBLEClient: Got Client event 
[17:24:29]I NimBLEClient: mtu update event; conn_handle=0 mtu=247
[17:24:29]I NimBLEClient: Connection established
[17:24:29]D NimBLEClient: >> deleteServices
[17:24:29]D NimBLEClient: Got Client event 
[17:24:29]D NimBLEClient: << deleteServices
[17:24:29]D [ 37749][D][NukiBle.cpp:1506] onConnect(): BLE connected
[17:24:29]NimBLEClient: Peer requesting to update connection parameters
[17:24:29]D NimBLEClient: MinInterval: 12, MaxInterval: 12, Latency: 0, Timeout: 600
[17:24:29]D NimBLEClient: << connect()
[17:24:29]D NimBLEClientCallbacks: onConnParamsUpdateRequest: default
[17:24:29]D NimBLEClient: >> getService: uuid: a92ee100-5501-11e4-916c-0800200c9a66
[17:24:29]D NimBLEClient: Accepted peer params
[17:24:29]D NimBLEClient: >> retrieveServices
[17:24:29]D NimBLEClient: Got Client event 
[17:24:29]I NimBLEClient: mtu update event; conn_handle=0 mtu=247
[17:24:29]D NimBLEClient: Service Discovered >> status: 0 handle: 23
[17:24:29]D NimBLERemoteService: >> NimBLERemoteService()
[17:24:29]D NimBLERemoteService: << NimBLERemoteService(): a92ee100-5501-11e4-916c-0800200c9a66
[17:24:29]D NimBLEClient: Service Discovered >> status: 14 handle: -1
[17:24:29]D NimBLEClient: << Service Discovered
[17:24:29]D NimBLEClient: << retrieveServices
[17:24:29]D NimBLERemoteService: >> getCharacteristic: uuid: a92ee101-5501-11e4-916c-0800200c9a66
[17:24:29]D NimBLERemoteService: >> retrieveCharacteristics() for service: a92ee100-5501-11e4-916c-0800200c9a66
[17:24:29]D NimBLEClient: Got Client event 
[17:24:29]I NimBLEClient: Connection parameters updated.
[17:24:29]D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 25
[17:24:29]D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
[17:24:29]D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): a92ee101-5501-11e4-916c-0800200c9a66
[17:24:29]D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
[17:24:29]D NimBLERemoteService: << Characteristic Discovered
[17:24:29]D NimBLERemoteService: << retrieveCharacteristics()
[17:24:29]D NimBLERemoteCharacteristic: >> setNotify(): Characteristic: uuid: a92ee101-5501-11e4-916c-0800200c9a66, handle: 25 0x0019, props:  0x28, 02
[17:24:29]D NimBLERemoteCharacteristic: >> getDescriptor: uuid: 0x2902
[17:24:29]D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: a92ee101-5501-11e4-916c-0800200c9a66
[17:24:29]D NimBLERemoteCharacteristic: Next Characteristic >> status: 14 handle: -1
[17:24:29]D NimBLERemoteCharacteristic: Descriptor Discovered >> status: 0 handle: 26
[17:24:29]D NimBLERemoteDescriptor: >> NimBLERemoteDescriptor()
[17:24:29]D NimBLERemoteDescriptor: << NimBLERemoteDescriptor(): 0x2902
[17:24:29]D NimBLERemoteCharacteristic: << Descriptor Discovered. status: 0
[17:24:29]D NimBLERemoteCharacteristic: << retrieveDescriptors(): Found 1 descriptors.
[17:24:29]D NimBLERemoteCharacteristic: << setNotify()
[17:24:29]D NimBLERemoteDescriptor: >> Descriptor writeValue: Descriptor: uuid: 0x2902, handle: 26
[17:24:29]I NimBLERemoteDescriptor: Write complete; status=0 conn_handle=0
[17:24:29]D NimBLERemoteDescriptor: << Descriptor writeValue, rc: 0
[17:24:29][ 38420][D][NukiBle.cpp:1242] registerOnGdioChar(): GDIO characteristic registered
[17:24:30]D NimBLEClient: >> getService: uuid: a92ee200-5501-11e4-916c-0800200c9a66
[17:24:30]D NimBLEClient: >> retrieveServices
[17:24:30]D NimBLEClient: Service Discovered >> status: 0 handle: 27
[17:24:30]D NimBLERemoteService: >> NimBLERemoteService()
[17:24:30]D NimBLERemoteService: << NimBLERemoteService(): a92ee200-5501-11e4-916c-0800200c9a66
[17:24:30]D NimBLEClient: Service Discovered >> status: 14 handle: -1
[17:24:30]D NimBLEClient: << Service Discovered
[17:24:30]D NimBLEClient: << retrieveServices
[17:24:30]D NimBLERemoteService: >> getCharacteristic: uuid: a92ee202-5501-11e4-916c-0800200c9a66
[17:24:30]D NimBLERemoteService: >> retrieveCharacteristics() for service: a92ee200-5501-11e4-916c-0800200c9a66
[17:24:30]D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 32
[... 49 lines cut ...]
[17:24:32][ 41071][D][NukiBle.cpp:1318] notifyCallback():  Notify callback for characteristic: a92ee101-5501-11e4-916c-0800200c9a66 of length: 36
[17:24:32][ 41083][D][NukiUtils.cpp:103] crcValid(): CRC CHECK OKE
[17:24:32][ 41091][D][NukiBle.cpp:1022] pairStateMachine(): ##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[17:24:32][ 41150][D][NukiBle.cpp:1039] pairStateMachine(): ##################### SEND AUTHENTICATOR #########################
[17:24:32]D NimBLEScan: >> stop()
[17:24:32]D NimBLEScan: << stop()
[17:24:32]D NimBLERemoteCharacteristic: >> writeValue(), length: 36
[17:24:32]I NimBLERemoteCharacteristic: Write complete; status=0 conn_handle=0
[... cut ...]
[17:24:33]E (49286) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[17:24:33]E (49286) task_wdt:  - loopTask (CPU 1)
[17:24:33]E (49286) task_wdt: Tasks currently running:
[17:24:33]E (49286) task_wdt: CPU 0: btController
[17:24:33]E (49286) task_wdt: CPU 1: IDLE
[17:24:33]E (49286) task_wdt: Aborting.
[17:24:33]
[17:24:33]abort() was called at PC 0x401035d8 on core 0
[17:24:33]
[17:24:34]
[17:24:34]Backtrace:0x40083701:0x3ffbeb1c |<-CORRUPTED
[17:24:34]
[17:24:34]
[17:24:34]
[17:24:34]
[17:24:34]ELF file SHA256: 0000000000000000

uriyacovy, 2024-08-12:
Thanks. 
1. Please post the full yaml.
2. Instead of using this library `  - https://github.com/uriyacovy/NukiBleEsp32` please try the original repo `https://github.com/I-Connect/NukiBleEsp32` and see if it works.

AntorFr, 2024-08-12:
Hi,

Thanks @uriyacovy  for your replay.
I was already using https://github.com/I-Connect/NukiBleEsp32 with the differentes releases / branches.

After a full day of testing I thing i find the root cause, and a work arround allowing me to pair my device.

So the issues is that Esphome has reduce the wst watchdog timer to 5s and any task taking more time causing a reset.
Nuki is taking too much time to answer to the "SEND AUTHENTICATOR" test causing watchdog reset.

Adding a "esp_task_wdt_reset();" in the loop to prevent the watchdog to trigger had allow me to successfully end the pairing process (tested several time to confirm)

you can find there my ugly workarround
https://github.com/I-Connect/NukiBleEsp32/commit/a1d9f9d5f7df09f84fb3c5659aebfb6a7ac42997

you can try it using 
`https://github.com/AntorFr/NukiBleEsp32#test` instead of `https://github.com/I-Connect/NukiBleEsp32`

uriyacovy, 2024-08-12:
Good catch! Thanks a lot!
I wonder why pairing takes longer in Nuki V4. 
In any case, maybe there's a workaround that can be done in this library. Can you check the following?
In `nuki_lock.cpp` add `#include "esphome/core/application.h"` and just before line 266 add `App.feed_wdt();`
Let me know if it works, and if it does, I'll add it to the code.

AntorFr, 2024-08-17:
Hi @uriyacovy 

Thanks, I tried it (you can look at the fix there https://github.com/AntorFr/ESPHome_nuki_lock/tree/fix-pairing-watchdog-reset) but a still had a watch dog reset.
I will try to find more time to investigate but for now I'm hosting some friends and family so do have much to do tecky stuff ^^

AzonInc, 2024-08-25:
I also use a v4 and solved it increasing the timeout by adding
`esp_task_wdt_init(15, false);` to `void NukiLockComponent::setup()`  a few months ago.

`App.feed_wdt();` was not enough.

thecode, 2024-10-26:
I think this can be closed as support for v4 was merged from @AzonInc

closed 2024-10-26 by uriyacovy