uriyacovy/ESPHome_nuki_lock #51: error in compiling with Esphome 2024.6.1

marcor78, 2024-06-24:
After updating Esphome to 2024.6.1, I can't update my Nuki Lock device and get errors during compiling.
Here the logs:
```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/nuki-lock.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing nuki-lock (board: esp32dev; framework: arduino; platform: platformio/espressif32@6.7.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- Preferences @ 2.0.0
|-- NimBLE-Arduino @ 2.0.0+sha.76ef4b7
|-- Crc16 @ 0.1.2+sha.719a7a1
|-- NukiBleEsp @ 2.0.0+sha.1de6290
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/nuki-lock/src/esphome/components/nuki_lock/nuki_lock.cpp.o
Compiling .pioenvs/nuki-lock/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/nuki-lock/src/esphome/components/sensor/filter.cpp.o
In file included from .piolibdeps/nuki-lock/NukiBleEsp/src/NukiLock.h:3,
                 from src/esphome/components/nuki_lock/nuki_lock.h:10,
                 from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
.piolibdeps/nuki-lock/NukiBleEsp/src/NukiBle.h:316:10: error: 'void Nuki::NukiBle::onDisconnect(NimBLEClient*)' marked 'override', but does not override
     void onDisconnect(BLEClient*) override;
          ^~~~~~~~~~~~
*** [.pioenvs/nuki-lock/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================= [FAILED] Took 24.02 seconds =========================
```

I already updated libraries as suggested for example [here ](https://github.com/uriyacovy/ESPHome_nuki_lock/issues/44), cleaned build files and re-install, but no joy so far.

This is my config:

```
esphome:
  name: nuki-lock
  libraries:
  - Preferences
  - https://github.com/h2zero/NimBLE-Arduino
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock
  
substitutions:
  hostname: 'Nuki Lock:'
  id: 'nuki-lock'
  
esp32:
  board: esp32dev  
  framework:
    type: arduino
    version: 2.0.16
    platform_version: 6.7.0
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret encryption
  reboot_timeout: 0s

ota:
  - platform: esphome
  
# WebServer and Authentication
web_server:
  port: 80
  auth:
    username: !secret web_server_username
    password: !secret web_server_password
  
# Setting time platform
time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Rome
    
# Wifi, AP and Static IP
wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_pwd
  - ssid: !secret 2nd_wifi_ssid
    password: !secret 2nd_wifi_pwd
  manual_ip:
    static_ip: 192.168.1.200
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nuki-Lock Fallback Hotspot"
    password: !secret hotspot_pwd

captive_portal:

lock:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
      
switch:
  - platform: restart
    name: $hostname Restart
    
```

ssorgatem, 2024-06-24:
Same issue here

Triskae, 2024-07-05:
Hello, same error here !
```
In file included from .piolibdeps/sensor-all-in-one-linvingroom/NukiBleEsp/src/NukiLock.h:3,
                 from src/esphome/components/nuki_lock/nuki_lock.h:10,
                 from src/esphome/components/nuki_lock/nuki_lock.cpp:2:
.piolibdeps/sensor-all-in-one-linvingroom/NukiBleEsp/src/NukiBle.h:316:10: error: 'void Nuki::NukiBle::onDisconnect(NimBLEClient*)' marked 'override', but does not override
     void onDisconnect(BLEClient*) override;
          ^~~~~~~~~~~~
*** [.pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
```

J8PMN, 2024-07-05:
same issue here

uriyacovy, 2024-07-10:
I'm still trying to figure out the change in ESPHome that cause this. Hopefully I could update in a few days.

Triskae, 2024-07-10:
Thanks a lot @uriyacovy

psxde, 2024-07-23:
It works if you include https://github.com/h2zero/NimBLE-Arduino#1.4.2 because in this library onDisconnect has changed from `void onDisconnect(NimBLEClient* pClient);` to `void onDisconnect(NimBLEClient* pClient, int reason);`

J8PMN, 2024-07-24:
Works for me! Thanks!

icygame, 2024-07-24:
and me, thanks!

Triskae, 2024-07-25:
I'll try this night, thank you !

UriYacovyAB, 2024-07-25:
Thanks @psxde!
Actually, the support NimBLE version is 1.4.0 according to 
https://github.com/I-Connect/NukiBleEsp32/blob/main/platformio.ini
Let me know if it works for you, and I'll update the documentation.

Triskae, 2024-07-25:
Not working here 😢 

```
src/esphome/components/api/api_pb2.h:875: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<int>()'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sensor-all-in-one-linvingroom/firmware.elf] Error 1
```

Triskae, 2024-07-29:
> > Not working here 😢
> > ```
> > src/esphome/components/api/api_pb2.h:875: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<int>()'
> > collect2: error: ld returned 1 exit status
> > *** [.pioenvs/sensor-all-in-one-linvingroom/firmware.elf] Error 1
> > ```
> 
> Not sure this is related to the same problem. Try `esphome clean` or delete `.esphome` folder and build again.

Nop :/ No changes even after clean ...

uriyacovy, 2024-07-30:
@Triskae, can you post your yaml and full logs?

Triskae, 2024-08-06:
Yep here it is

```
esphome:
  name: sensor-all-in-one-linvingroom
  friendly_name: sensor-all-in-one-linvingroom
  libraries:
  - Preferences
  - https://github.com/h2zero/NimBLE-Arduino#1.3.8
  - Crc16
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  - platform: esphome
    password: !secret ota_password

safe_mode:
  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sensor-all-in-one-linvingroom"
    password: "redacted"

captive_portal:

uart:
  rx_pin: GPIO5
  baud_rate: 9600

i2c:
  sda: GPIO6
  scl: GPIO7
  scan: false

sensor:
  - platform: pm1006
    pm_2_5:
      name: "Particulate Matter 2.5µm Concentration"
  - platform: scd4x
    id: co2
    co2:
      name: "Living Room CO2"
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    automatic_self_calibration: False
    temperature_offset: 7
[... 50 lines cut ...]
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
Linking .pioenvs/sensor-all-in-one-linvingroom/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o: in function `void esphome::api::UserServiceBase<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int>::execute_<0, 1>(std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> >, esphome::seq<0, 1>)':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/user_services.h:57: undefined reference to `int esphome::api::get_execute_arg_value<int>(esphome::api::ExecuteServiceArgument const&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o: in function `void esphome::api::UserServiceBase<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, bool>::execute_<0, 1, 2, 3>(std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> >, esphome::seq<0, 1, 2, 3>)':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/user_services.h:57: undefined reference to `int esphome::api::get_execute_arg_value<int>(esphome::api::ExecuteServiceArgument const&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/user_services.h:57: undefined reference to `int esphome::api::get_execute_arg_value<int>(esphome::api::ExecuteServiceArgument const&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o: in function `void esphome::api::UserServiceBase<int>::execute_<0>(std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> >, esphome::seq<0>)':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/user_services.h:57: undefined reference to `int esphome::api::get_execute_arg_value<int>(esphome::api::ExecuteServiceArgument const&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o: in function `esphome::api::UserServiceBase<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int>::encode_list_service_response()':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/user_services.h:35: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<int>()'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o: in function `esphome::api::ListEntitiesServicesArgument::ListEntitiesServicesArgument()':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/api_pb2.h:875: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<int>()'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sensor-all-in-one-linvingroom/src/esphome/components/nuki_lock/nuki_lock.cpp.o: in function `esphome::api::UserServiceBase<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, bool>::encode_list_service_response()':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/user_services.h:35: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<int>()'
[... cut ...] ion `esphome::api::ListEntitiesServicesArgument::ListEntitiesServicesArgument()':
/data/build/sensor-all-in-one-linvingroom/src/esphome/components/api/api_pb2.h:875: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<int>()'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sensor-all-in-one-linvingroom/firmware.elf] Error 1
========================= [FAILED] Took 11.02 seconds =========================

```

Tell me if you need something more 👍🏼

Triskae, 2024-08-06:
I tested on a brand new project, and worked like a charm, so it's something to do with the other I2C device, or the board itself....

uriyacovy, 2024-08-06:
Can you post the full yaml?

Triskae, 2024-08-06:
> Can you post the full yaml?

The full yaml is just above

uriyacovy, 2024-08-07:
Try to add the following below `esp32`:
```
  framework:
    type: arduino
    version: 2.0.16
    platform_version: 6.7.0

```

Triskae, 2024-08-07:
> Try to add the following below `esp32`:
> 
> ```
>   framework:
>     type: arduino
>     version: 2.0.16
>     platform_version: 6.7.0
> ```

Nop tried right now, and stiil not working ... Even after clean.... really strange ... Working without I2C and UART, but not working with those ...

Triskae, 2024-08-08:
> I was able to reproduce and it seems to be related to the c3 board you are using. See in this thread how to fix (worked for me): [esphome/issues#3564 (comment)](https://github.com/esphome/issues/issues/3564#issuecomment-1821603239)

Oooooh nice !! working now !! thank you very much !!

closed 2024-08-08 by uriyacovy