uriyacovy/ESPHome_nuki_lock #57: Problems compiling

pkvapil, 2024-10-23:
I set this up few weeks back now when i wanted to add ble proxy i cant compile it (even without the proxy stuff)
```
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:18:4: error: #error "For ESP32 devices, please use the native Preferences library"
   #error "For ESP32 devices, please use the native Preferences library"
    ^~~~~
Compiling .pioenvs/nukicat/libf1b/NimBLE-Arduino@src-bd1d552eb48cf957bbb0ac4df52c4fce/NimBLE2904.cpp.o
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'bool Preferences::begin(const char*, bool)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:89:14: error: '_fs_init' was not declared in this scope
         if (!_fs_init()) {
              ^~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:89:14: note: suggested alternative: '__sinit'
         if (!_fs_init()) {
              ^~~~~~~~
              __sinit
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:93:14: error: '_fs_mkdir' was not declared in this scope
         if (!_fs_mkdir(NVS_PATH)) {
              ^~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:93:14: note: suggested alternative: '_fseeko_r'
         if (!_fs_mkdir(NVS_PATH)) {
              ^~~~~~~~~
              _fseeko_r
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:109:9: error: '_fs_mkdir' was not declared in this scope
     if (_fs_mkdir(p.c_str())) {
         ^~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:109:9: note: suggested alternative: '_fseeko_r'
     if (_fs_mkdir(p.c_str())) {
         ^~~~~~~~~
         _fseeko_r
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'bool Preferences::clear()':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:144:9: error: '_fs_clean_dir' was not declared in this scope
     if (_fs_clean_dir(_path.c_str())) {
         ^~~~~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:144:9: note: suggested alternative: '_fscanf_r'
     if (_fs_clean_dir(_path.c_str())) {
         ^~~~~~~~~~~~~
         _fscanf_r
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:146:16: error: '_fs_mkdir' was not declared in this scope
         return _fs_mkdir(p.c_str());
                ^~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:146:16: note: suggested alternative: '_fseeko_r'
         return _fs_mkdir(p.c_str());
                ^~~~~~~~~
                _fseeko_r
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'bool Preferences::remove(const char*)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:161:12: error: '_fs_unlink' was not declared in this scope
     return _fs_unlink(path.c_str());
            ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:161:12: note: suggested alternative: 'unlink'
     return _fs_unlink(path.c_str());
            ^~~~~~~~~~
            unlink
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'size_t Preferences::putBytes(const char*, const void*, size_t)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:235:9: error: '_fs_exists' was not declared in this scope
     if (_fs_exists(path.c_str())) {
         ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:235:9: note: suggested alternative: 'on_exit'
     if (_fs_exists(path.c_str())) {
         ^~~~~~~~~~
         on_exit
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:239:13: error: '_fs_verify' was not declared in this scope
         if (_fs_verify(path.c_str(), buf, len)) {
             ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:246:23: error: '_fs_create' was not declared in this scope
         int written = _fs_create(next.c_str(), buf, len);
                       ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:246:23: note: suggested alternative: 'timer_create'
         int written = _fs_create(next.c_str(), buf, len);
                       ^~~~~~~~~~
                       timer_create
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:248:13: error: '_fs_rename' was not declared in this scope
         if (_fs_rename(next.c_str(), path.c_str())) {
             ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:248:13: note: suggested alternative: 'basename'
         if (_fs_rename(next.c_str(), path.c_str())) {
             ^~~~~~~~~~
             basename
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:255:16: error: '_fs_create' was not declared in this scope
         return _fs_create(path.c_str(), buf, len);
                ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:255:16: note: suggested alternative: 'timer_create'
         return _fs_create(path.c_str(), buf, len);
                ^~~~~~~~~~
                timer_create
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'bool Preferences::isKey(const char*)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:270:12: error: '_fs_exists' was not declared in this scope
     return _fs_exists(path.c_str());
            ^~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:270:12: note: suggested alternative: 'on_exit'
     return _fs_exists(path.c_str());
            ^~~~~~~~~~
            on_exit
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'String Preferences::getString(const char*, String)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:365:15: error: '_fs_get_size' was not declared in this scope
     int len = _fs_get_size(path.c_str());
               ^~~~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:365:15: note: suggested alternative: '__getline'
     int len = _fs_get_size(path.c_str());
               ^~~~~~~~~~~~
               __getline
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'size_t Preferences::getBytesLength(const char*)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:384:15: error: '_fs_get_size' was not declared in this scope
     int len = _fs_get_size(path.c_str());
               ^~~~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:384:15: note: suggested alternative: '__getline'
     int len = _fs_get_size(path.c_str());
               ^~~~~~~~~~~~
               __getline
.piolibdeps/nukicat/Preferences/src/Preferences.cpp: In member function 'size_t Preferences::getBytes(const char*, void*, size_t)':
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:394:15: error: '_fs_get_size' was not declared in this scope
     int len = _fs_get_size(path.c_str());
               ^~~~~~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:394:15: note: suggested alternative: '__getline'
     int len = _fs_get_size(path.c_str());
               ^~~~~~~~~~~~
               __getline
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:408:12: error: '_fs_read' was not declared in this scope
     return _fs_read(path.c_str(), buf, len);
            ^~~~~~~~
.piolibdeps/nukicat/Preferences/src/Preferences.cpp:408:12: note: suggested alternative: 'fread'
     return _fs_read(path.c_str(), buf, len);
            ^~~~~~~~
            fread
*** [.pioenvs/nukicat/lib95b/Preferences/Preferences.cpp.o] Error 1
```

```
esphome:
  name: nukicat
  friendly_name: NukiCat
  libraries:
  - Preferences
  - https://github.com/h2zero/NimBLE-Arduino#1.4.0
  - Crc16
  - https://github.com/uriyacovy/NukiBleEsp32
external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock
esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.16
    platform_version: 6.7.0 

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nukicat Fallback Hotspot"
    password: "XXX"

captive_portal:

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki Lock
    is_connected: 
      name: "Nuki Connected"
    is_paired: 
      name: "Nuki Paired"      
  # Optional:
    battery_critical:
      name: "Nuki Battery Critical"
    battery_level:
      name: "Nuki Battery Level"
    door_sensor:
      name: "Nuki Door Sensor"
    door_sensor_state:
      name: "Nuki Door Sensor State"

    
```

uriyacovy, 2024-10-23:
Seems like a build issue. Try running `esphome clean`

alexflcl, 2024-12-14:
I had same issue and solved adding @dev to the following lines:

esphome:
  name: nukibridge
  friendly_name: Nuki Bridge
  libraries:
  - Preferences
  - https://github.com/h2zero/NimBLE-Arduino#1.4.2
  - Crc16
  **- https://github.com/uriyacovy/NukiBleEsp32#dev**

external_components:
  **- source: github://uriyacovy/ESPHome_nuki_lock@dev**

I hope this works for you :)

AzonInc, 2024-12-14:
With the new version you don't need to add libraries manually anymore.

Just remove everything of this block;

```
libraries:

Preferences
https://github.com/h2zero/NimBLE-Arduino#1.4.2
Crc16
- https://github.com/uriyacovy/NukiBleEsp32#dev
```

alexflcl, 2024-12-14:
> With the new version you don't need to add libraries manually anymore.
> 
> Just remove everything of this block;
> 
> ```
> libraries:
> 
> - Preferences
> https://github.com/h2zero/NimBLE-Arduino#1.4.2
> - Crc16
> - https://github.com/uriyacovy/NukiBleEsp32#dev
> ```

Nice, it's working! Thank you :D

closed 2024-12-16 by uriyacovy