uriyacovy/ESPHome_nuki_lock #65: 'NimBLEServer' in namespace 'esphome::esp32_ble_server' does not name a type

mrad86, 2024-11-18:
Hi,

I am getting following compile error:

> Compiling .pioenvs/esphome-web-a35c24/lib64d/WiFi/WiFiMulti.cpp.o
> In file included from .piolibdeps/esphome-web-a35c24/NukiBleEsp/src/NukiBle.h:14,
>                  from .piolibdeps/esphome-web-a35c24/NukiBleEsp/src/NukiLock.h:3,
>                  from src/esphome/components/nuki_lock/nuki_lock.h:33,
>                  from src/esphome.h:50,
>                  from src/main.cpp:3:
> .piolibdeps/esphome-web-a35c24/NimBLE-Arduino/src/NimBLEDevice.h:68:41: error: 'NimBLEServer' in namespace 'esphome::esp32_ble_server' does not name a type
>  #define BLEServer                       NimBLEServer
>                                          ^~~~~~~~~~~~
> .piolibdeps/esphome-web-a35c24/NimBLE-Arduino/src/NimBLEDevice.h:68:41: note: in definition of macro 'BLEServer'
>  #define BLEServer                       NimBLEServer
>                                          ^~~~~~~~~~~~
> src/main.cpp:76:1: note: suggested alternative: 'BLEServer'
>  esp32_ble_server::BLEServer *esp32_ble_server_bleserver_id;
>  ^~~~~~~~~~~~~~~~
>  BLEServer
> /config/esphome/esphome-web-a35c24.yaml: In function 'void setup()':
> /config/esphome/esphome-web-a35c24.yaml:160:3: error: 'esp32_ble_server_bleserver_id' was not declared in this scope
> /config/esphome/esphome-web-a35c24.yaml:160:3: note: suggested alternative: 'web_server_webserver_id'
> /config/esphome/esphome-web-a35c24.yaml:160:39: error: expected type-specifier
> Compiling .pioenvs/esphome-web-a35c24/lib64d/WiFi/WiFiSTA.cpp.o
> Compiling .pioenvs/esphome-web-a35c24/lib64d/WiFi/WiFiScan.cpp.o
> *** [.pioenvs/esphome-web-a35c24/src/main.cpp.o] Error 1
> ========================= [FAILED] Took 127.25 seconds =========================

I am using a ESP32-WROOM-32 with following yaml:
```
substitutions:
  name: esphome-web-a35c24
  friendly_name: Nuki Lock

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esp32:
  board: "esp32dev"  # Or whatever other board you're using
  framework:
    type: arduino
    version: 2.0.16
    platform_version: 6.7.0

lock:
  # Required
  - platform: nuki_lock
    name: Nuki Lock
  # Optional: Settings
    pairing_mode_timeout: 300s
    event: "nuki"
  # Optional: Binary Sensors
    is_connected:
      name: "Nuki Connected"
    is_paired:
      name: "Nuki Paired"
    battery_critical:
      name: "Nuki Battery Critical"
    door_sensor:
      name: "Nuki Door Sensor"
  # Optional: Sensors
    battery_level:
      name: "Nuki Battery Level"
  # Optional: Text Sensors
    door_sensor_state:
      name: "Nuki Door Sensor: State"
    last_unlock_user:
      name: "Nuki Last Unlock User"
    last_lock_action:
      name: "Nuki Last Lock Action"
    last_lock_action_trigger:
      name: "Nuki Last Lock Action Trigger"
  # Optional: Switches
    pairing_mode:
      name: "Nuki Pairing Mode"
    auto_unlatch:
      name: "Nuki Auto unlatch"
    button_enabled:
      name: "Nuki Button: Locking operations"
    led_enabled:
      name: "Nuki LED Signal"
  # Optional: Number Inputs
    security_pin:
      name: "Nuki Security Pin"
    led_brightness:
      name: "Nuki LED Brightness"
    timezone_offset:
      name: "Nuki Timezone: Offset"
  # Optional: Switches
    night_mode_enabled:
      name: "Nuki Night Mode"
    night_mode_auto_lock_enabled:
      name: "Nuki Night Mode: Auto Lock"
    night_mode_auto_unlock_disabled:
      name: "Nuki Night Mode: Reject Auto Unlock"
    night_mode_immediate_lock_on_start_enabled:
      name: "Nuki Night Mode: Lock at Start Time"
    auto_lock_enabled:
      name: "Nuki Auto Lock"
    auto_unlock_disabled:
      name: "Nuki Auto Unlock: Disable"
    immediate_auto_lock_enabled:
      name: "Nuki Auto Lock: Immediately"
    auto_update_enabled:
      name: "Nuki Automatic Updates"
    single_lock_enabled:
      name: "Nuki Single Lock"
    dst_mode_enabled:
      name: "Nuki Daylight Saving Time"
  # Optional: Select Inputs
    single_buton_press_action:
      name: "Nuki Button: Single Press Action"
    double_buton_press_action:
      name: "Nuki Button: Double Press Action"
    fob_action_1:
      name: "Nuki Fob: Action 1"
    fob_action_2:
      name: "Nuki Fob: Action 2"
    fob_action_3:
      name: "Nuki Fob: Action 3"
    timezone:
      name: "Nuki Timezone"
    advertising_mode:
      name: "Nuki Advertising Mode"
  # Optional: Buttons
    unpair:
      name: "Nuki Unpair Device"
  # Optional: Callbacks
    on_pairing_mode_on_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned on");
    on_pairing_mode_off_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned off");
    on_paired_action:
      - lambda: ESP_LOGI("nuki_lock", "Paired sucessfuly");
```

Thanks for the support!

AzonInc, 2024-11-18:
Hi,

I see u are using Improv via BLE which doesn't work well with NimBLE.

Try removing `esp32_improv` and compile again. That should allow you to compile ESPHome.

mrad86, 2024-11-19:
that solved the issue. Thank you!

closed 2024-11-19 by mrad86