uriyacovy/ESPHome_nuki_lock #66: API diconnects every second (approx. since V 0.0.4)

Pe-MaKer, 2024-12-16:
Every few days, my ESPHome-Nuki-Lock enters a state where the api conntection to HA becomes unstable.
This never occured on older versions of this outstanding integration.

See below the HA Core Logs and a screenshot of the ESP32 Log.

```
Logger: aioesphomeapi.connection
Quelle: runner.py:154
Erstmals aufgetreten: 10:09:34 (2015 Vorkommnisse)
Zuletzt protokolliert: 10:45:58

haustuer @ 10.0.2.3: Connection error occurred: Invalid protobuf message: type=HomeassistantServiceResponse data=b'\n\x0cesphome.nuki\x12\x14\n\x06action\x12\nDoorClosed\x12\x1d\n\x0fauthorizationId\x12\n4294967295\x12;\n\x11authorizationName\x12&\xfd\xdb${\xaa\xc8\xe4DN\xec@\n\xc4A\xe5&\x0f\xca%&\xf8a\xd8\xf09?\xd1\xc3,\x96\x06\xfe\xc2\xf2_+\x01\xc4\x12\x0e\n\x05index\x12\x0528210\x12\r\n\x07timeDay\x12\x0215\x12\x0e\n\x08timeHour\x12\x0211\x12\x0f\n\ntimeMinute\x12\x016\x12\x0f\n\ttimeMonth\x12\x0212\x12\x10\n\ntimeSecond\x12\x0243\x12\x10\n\x08timeYear\x12\x042024\x12\x12\n\x04type\x12\nDoorSensor(\x01': Error parsing message with type 'HomeassistantServiceResponse'
haustuer @ 10.0.2.3: Connection error occurred: Invalid protobuf message: type=TextSensorStateResponse data=b'\rF\x13j^\x12&\xfd\xdb${\xaa\xc8\xe4DN\xec@\n\xc4A\xe5&\x0f\xca%&\xf8a\xd8\xf09?\xd1\xc3,\x96\x06\xfe\xc2\xf2_+\x01\xc4': Error parsing message with type 'TextSensorStateResponse'
haustuer @ 10.0.2.3: Connection error occurred: Invalid protobuf message: type=HomeassistantServiceResponse data=b'\n\x0cesphome.nuki\x12\x14\n\x06action\x12\nDoorClosed\x12\x1d\n\x0fauthorizationId\x12\n4294967295\x123\n\x11authorizationName\x12\x1e\xfd\xdb${\xaa\xc8\xe4DN\xec@\n\xc4A\xe5&\x0f\xca%&\xf8a\xd8\xf09?\xd1\xc3=k\x12\x0e\n\x05index\x12\x0528210\x12\r\n\x07timeDay\x12\x0215\x12\x0e\n\x08timeHour\x12\x0211\x12\x0f\n\ntimeMinute\x12\x016\x12\x0f\n\ttimeMonth\x12\x0212\x12\x10\n\ntimeSecond\x12\x0243\x12\x10\n\x08timeYear\x12\x042024\x12\x12\n\x04type\x12\nDoorSensor(\x01': Error parsing message with type 'HomeassistantServiceResponse'
haustuer @ 10.0.2.3: Connection error occurred: Invalid protobuf message: type=TextSensorStateResponse data=b'\rF\x13j^\x12\x1e\xfd\xdb${\xaa\xc8\xe4DN\xec@\n\xc4A\xe5&\x0f\xca%&\xf8a\xd8\xf09?\xd1\xc3=k': Error parsing message with type 'TextSensorStateResponse'
```
[image]

AzonInc, 2024-12-17:
Hi,

Could you please try version 0.0.4 and confirm whether the issue starts with this version? There are no significant changes between 0.0.3 and 0.0.4, so it would be helpful to verify this.

Thanks

Pe-MaKer, 2024-12-21:
@AzonInc I have no clue on how to use a specific version.
I need to clear the built files, I assume. But how can I compile using a previous version?

uriyacovy, 2024-12-22:
Use the following:
```
external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@v0.0.4

```

Pe-MaKer, 2024-12-27:
> Use the following:
> 
> ```
> external_components:
>   - source: github://uriyacovy/ESPHome_nuki_lock@v0.0.4
> ```

It's throwing a fatal error - even after commenting out the recent added sensor like nightmode and last_unlock_user:
```
Compiling .pioenvs/haustuer/src/esphome/components/nuki_lock/nuki_lock.cpp.o
In file included from src/esphome/components/nuki_lock/nuki_lock.cpp:3:
src/esphome/components/nuki_lock/nuki_lock.h:12:10: fatal error: NukiLock.h: No such file or directory

******************************************************************
* Looking for NukiLock.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NukiLock.h"
* Web  > https://registry.platformio.org/search?q=header:NukiLock.h
*
******************************************************************

 #include "NukiLock.h"
          ^~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/haustuer/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================= [FAILED] Took 25.96 seconds =========================```

AzonInc, 2024-12-28:
Did u add the libraries block? It's still needed for versions < 0.0.5

```
esphome:
  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32
```

Pe-MaKer, 2025-01-07:
Thanks @AzonInc 
But I still get this.
[image]

Here is my code:
```
substitutions:
  display_name: haustuer

  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password

# ------------------------------
# ------------------------------
esphome:
  name: ${display_name}
  friendly_name: Haustür

  libraries:
  - Preferences
  - https://github.com/vinmenn/Crc16.git
  - https://github.com/uriyacovy/NukiBleEsp32

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@v0.0.4
  #- source: github://Pe-MaKer/ESPHome_nuki_lock@dev

esp32:
  board: "wemos_d1_mini32"
  framework:
    type: arduino
    version: 2.0.16
    platform_version: 6.7.0

wifi:
  networks:
  - ssid: ${wifi_ssid}
    password: ${wifi_password}
  - ssid: ${wifi_ssid2}
    password: ${wifi_password2}
  # Optional manual IP
  manual_ip:
    static_ip: 10.0.X.XX
    gateway: 10.0.0.1
    subnet: 255.0.0.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${display_name}_AP
    password: ${wifi_password}
    ap_timeout: 1min

captive_portal:

#bluetooth_proxy:
#  active: true

logger:
  level: DEBUG
api:

#ota:
#  password: "notwhatyouthink"
ota:
  - platform: esphome    
    password: !secret ota_password
    id: my_ota
safe_mode:

web_server:
  port: 80
  include_internal: true

number:
  - platform: template
    name: "Türöffner Delay"
    id: turoffner_delay
    optimistic: true
    min_value: 0      # Minimale Dauer des Delay in Sekunden
    max_value: 10000     # Maximale Dauer des Delay in Sekunden
    step: 250
    initial_value: 2000  # Default-Wert für den Delay beim Start
  - platform: template
    name: "Durchgangslicht Delay"
    id: durchgangslicht_delay
    optimistic: true
    min_value: 1      # Minimale Dauer des Delay in Sekunden
    max_value: 600     # Maximale Dauer des Delay in Sekunden
    step: 1
    initial_value: 120  # Default-Wert für den Delay beim Start

button:
  - platform: template
    name: "Türöffner"
    id: turoffner
    icon: "mdi:door-open"
    on_press:
      - if:
          condition: 
            lock.is_locked: nuki
          then:
            - lock.unlock: nuki
            - delay: 5s
            - switch.turn_on: oeffner
            - delay: !lambda "return id(turoffner_delay).state;"
            - switch.turn_off: oeffner

          else:
            - switch.turn_on: oeffner
            - delay: !lambda "return id(turoffner_delay).state;"
            #- delay: 2000ms
            - switch.turn_off: oeffner
  - platform: template
    name: "Durchgangslicht"
    id: durchgangslicht
    icon: "mdi:wall-sconce-flat"
    on_press:
      - output.turn_on: licht
      #- delay: 120000ms
      - delay: !lambda "return 1000 * id(durchgangslicht_delay).state;"
      - output.turn_off: licht

  - platform: restart
    name: "Restart haustuer"
  
switch:
  - platform: gpio
    id: oeffner
    pin: 16
    restore_mode: ALWAYS_OFF
    name: "Schnapper"
    icon: "mdi:gate"

output:
  - platform: gpio
    id: licht
    pin: 17    

    
light:
  - platform: binary
    name: "Licht"
    output: licht

lock:
  # Required:
  - platform: nuki_lock
    name: Nuki ESP
    id: nuki
    is_connected: 
      name: "Nuki ESP verbunden"
    is_paired: 
      name: "Nuki ESP gekoppelt"      

    battery_critical:
      name: "Nuki ESP Batterie kritisch"
    battery_level:
      name: "Nuki ESP Batteriestand"
    door_sensor:
      name: "Nuki ESP Türsensor"
    door_sensor_state:
      name: "Nuki ESP Türkontakt Zustand"
    unpair:
      name: "Nuki Unpair"
    pairing_mode:
      name: "Nuki Pairing Mode"

    on_pairing_mode_on_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing Mode aktiviert");
    on_pairing_mode_off_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing Mode deaktiviert");
    on_paired_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing erfolgreich");

```

AzonInc, 2025-01-07:
Did u also clean the build files?

Pe-MaKer, 2025-01-07:
> Did u also clean the build files?

Yes, I did.
Here is the full log to the code above after cleaning the build files:

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/tuer.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing haustuer (board: wemos_d1_mini32; framework: arduino; platform: platformio/espressif32@6.7.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.4
INFO Installing esphome/AsyncTCP-esphome @ 2.1.4
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
INFO AsyncTCP-esphome@2.1.4 has been installed!
Library Manager: Installing git+https://github.com/vinmenn/Crc16.git
INFO Installing git+https://github.com/vinmenn/Crc16.git
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-je_dbp2c'...
Library Manager: Crc16@0.1.2+sha.719a7a1 has been installed!
INFO Crc16@0.1.2+sha.719a7a1 has been installed!
Library Manager: Installing git+https://github.com/uriyacovy/NukiBleEsp32
INFO Installing git+https://github.com/uriyacovy/NukiBleEsp32
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-6pth0h0n'...
Library Manager: NukiBleEsp@2.0.0+sha.93e7da9 has been installed!
INFO NukiBleEsp@2.0.0+sha.93e7da9 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing h2zero/NimBLE-Arduino @ ^1.4.0
INFO Installing h2zero/NimBLE-Arduino @ ^1.4.0
Unpacking  [####################################]  100%
Library Manager: NimBLE-Arduino@1.4.3 has been installed!
INFO NimBLE-Arduino@1.4.3 has been installed!
Library Manager: Installing git+https://github.com/vinmenn/Crc16
INFO Installing git+https://github.com/vinmenn/Crc16
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-kbz7cnaq'...
Library Manager: CRC16@0.1.2+sha.719a7a1 has been installed!
INFO CRC16@0.1.2+sha.719a7a1 has been installed!
Library Manager: Installing git+https://github.com/I-Connect/BleScanner
INFO Installing git+https://github.com/I-Connect/BleScanner
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-_8lsvacy'...
Library Manager: BleScanner@1.1.0+sha.2b6c045 has been installed!
INFO BleScanner@1.1.0+sha.2b6c045 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
INFO ESPAsyncWebServer-esphome@3.2.2 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- Preferences @ 2.0.0
|-- Crc16 @ 0.1.2+sha.719a7a1
[... 49 lines cut ...]
Compiling .pioenvs/haustuer/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/haustuer/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/haustuer/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/haustuer/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/haustuer/src/esphome/components/nuki_lock/nuki_lock.cpp.o
Compiling .pioenvs/haustuer/src/esphome/components/number/automation.cpp.o
In file included from .piolibdeps/haustuer/NukiBleEsp/src/NukiLock.h:3,
                 from src/esphome/components/nuki_lock/nuki_lock.h:12,
                 from src/esphome/components/nuki_lock/nuki_lock.cpp:3:
.piolibdeps/haustuer/NukiBleEsp/src/NukiBle.h:14:10: fatal error: NimBLEDevice.h: No such file or directory

**********************************************************************
* Looking for NimBLEDevice.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NimBLEDevice.h"
* Web  > https://registry.platformio.org/search?q=header:NimBLEDevice.h
*
**********************************************************************

 #include "NimBLEDevice.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/haustuer/src/esphome/components/nuki_lock/nuki_lock.cpp.o] Error 1
========================= [FAILED] Took 30.49 seconds =========================
```

Pe-MaKer, 2025-01-08:
Thanks @AzonInc 
that did the trick. 
[image]
I will let you know for sure, if the issue occurs on v0.0.4 as well.

Let me know if there is anything that I can contribute to narrow this issue down. It would be great to have all the recently added sensors and switches, but at the house door, reliability rules.
I'm using the ESPHome Nuki component since it's release and it was released and it was always reliable. But since my wife was standing in front of the locked door last autumn, the "WAF" dropped significantly.

Pe-MaKer, 2025-01-09:
It would be nice to have such a flag in the current version of the component. Enable requests for debugging purposes in a "Smart Home Maintenance window" and disable them when the family needs a reliable door lock.

In fact the Last Unlock User sensor was always on my wishlist since it is supported by the Nuki-App itself as well as by the ESP32 nuki_hub project. So if there is anything I can provide (eg. Event Logs) just let me know.

At the moment, version 0.0.4 is up and running since yesterday.

Pe-MaKer, 2025-01-21:
The Node crashed again:
`Can't connect to ESPHome API for haustuer @ 10.0.2.3: Timeout waiting for HelloResponse, ConnectResponse after 30.0s (TimeoutAPIError)`

[image]

AzonInc, 2025-01-21:
And you're still on 0.0.4?

Pe-MaKer, 2025-01-21:
I would try v0.0.3 next because it's the last known good version in my personal setup.
Just to sort out hardware related issues, I replaced the ESP in my project.

[image]

AzonInc, 2025-01-21:
> I would try v0.0.3 next

Okay please try that and keep us updated.

AzonInc, 2025-02-06:
There is a new version based on IDF. Would you like to try it without the Arduino framework?  
This version uses a different approach of actively requesting data from the lock.  

You can find more information here: #72

AzonInc, 2025-11-24:
It's been almost a year now - I will close this for now.
Feel free to open a new issue or reopen this if the problem persists.

closed 2025-11-24 by AzonInc