uriyacovy/ESPHome_nuki_lock #67: Support for Smart Lock Ultra

B-Hartley, 2025-01-02:
How practical is it for you to modify (or guide me how to modify) this to support Smart Lock Ultra?
Is there anything I can do to give you the info you would need?

Thanks

AzonInc, 2025-01-07:
This component is based on the https://github.com/I-Connect/NukiBleEsp32 project so you should open an issue there I guess.

If it's supported by that project, it's possible to support it in this component as well.

B-Hartley, 2025-01-07:
Thanks for replying.
I realised that my first error was probably not specifying the security pin to allow pairing to work.
Unfortunately I can't.
The Ultra requires a 6 digit pin and as the PIN field is a "number" it won't allow numbers that big.
It would need to be a text field.

I guess I'll have to fork and have an experiment changing it to a text field so I can put in a bigger number.

AzonInc, 2025-01-07:
The docs of Nuki state that it's a uint16 which has a max of 65535 but thats as of 2020.
It could have changed now.

B-Hartley, 2025-01-07:
[image]

Yes. It requires a 6 digit pin. So guess it's either a 32 or text.

AzonInc, 2025-01-13:
We should open an issue in the other project for now so they add support for it.

AzonInc, 2025-02-12:
If you're still down for testing the ultra version it's here: https://github.com/uriyacovy/ESPHome_nuki_lock/tree/ultra
This one also supports esp-idf which I would recommend in general.

Example config:
```yaml
external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@ultra

esp32:
  board: "esp32dev"
  framework:
    type: esp-idf

lock:
  # Required
  - platform: nuki_lock
    name: Nuki Lock
  # Optional: Settings
    pairing_mode_timeout: 300s
    event: "nuki"
    security_pin: 1234
  # Optional: Advanced Settings
    alternative_connect_mode: true
    pairing_as_app: false
    ultra_pairing_mode: false
    query_interval_config: 3600s
    query_interval_auth_data: 7200s

  # Optional: Binary Sensors
    is_connected:
      name: "Nuki Connected"
    is_paired:
      name: "Nuki Paired"
    battery_critical:
      name: "Nuki Battery Critical"
    door_sensor:
      name: "Nuki Door Sensor"
  # Optional: Sensors
    battery_level:
      name: "Nuki Battery Level"
    bt_signal_strength:
      name: "Bluetooth Signal Strength"
  # Optional: Text Sensors
    door_sensor_state:
      name: "Nuki Door Sensor: State"
    last_unlock_user:
      name: "Nuki Last Unlock User"
    last_lock_action:
      name: "Nuki Last Lock Action"
    last_lock_action_trigger:
      name: "Nuki Last Lock Action Trigger"
    pin_status:
      name: "Nuki Security Pin Status"
  # Optional: Number Inputs
    led_brightness:
      name: "Nuki LED Brightness"
    timezone_offset:
      name: "Nuki Timezone: Offset"
    lock_n_go_timeout:
      name: "Nuki LockNGo Timeout"
  # Optional: Switches
    pairing_mode:
      name: "Nuki Pairing Mode"
    auto_unlatch:
      name: "Nuki Auto unlatch"
    button_enabled:
      name: "Nuki Button: Locking operations"
    led_enabled:
      name: "Nuki LED Signal"
    night_mode_enabled:
      name: "Nuki Night Mode"
    night_mode_auto_lock_enabled:
      name: "Nuki Night Mode: Auto Lock"
    night_mode_auto_unlock_disabled:
      name: "Nuki Night Mode: Reject Auto Unlock"
    night_mode_immediate_lock_on_start_enabled:
      name: "Nuki Night Mode: Lock at Start Time"
    auto_lock_enabled:
      name: "Nuki Auto Lock"
    auto_unlock_disabled:
      name: "Nuki Auto Unlock: Disable"
    immediate_auto_lock_enabled:
      name: "Nuki Auto Lock: Immediately"
    auto_update_enabled:
      name: "Nuki Automatic Updates"
    single_lock_enabled:
      name: "Nuki Single Lock"
    dst_mode_enabled:
      name: "Nuki Daylight Saving Time"
    auto_battery_type_detection_enabled:
      name: "Nuki Automatic Battery Type Detection"
    slow_speed_during_night_mode_enabled:
      name: "Nuki Slow Speed During Night Mode"
  # Optional: Select Inputs
    single_buton_press_action:
      name: "Nuki Button: Single Press Action"
    double_buton_press_action:
      name: "Nuki Button: Double Press Action"
    fob_action_1:
      name: "Nuki Fob: Action 1"
    fob_action_2:
      name: "Nuki Fob: Action 2"
    fob_action_3:
      name: "Nuki Fob: Action 3"
    timezone:
      name: "Nuki Timezone"
    advertising_mode:
      name: "Nuki Advertising Mode"
    battery_type:
      name: "Nuki Battery Type"
    motor_speed:
      name: "Nuki Motor Speed"
  # Optional: Buttons
    unpair:
      name: "Nuki Unpair Device"
  # Optional: Callbacks
    on_pairing_mode_on_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned on");
    on_pairing_mode_off_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned off");
    on_paired_action:
      - lambda: ESP_LOGI("nuki_lock", "Paired sucessfuly");
```

philipparndt, 2025-02-26:
I can confirm this is working with the Ultra Lock.
Maybe it is worth noting, that you need to enable Bluetooth pairing in the Nuki App to make this work (took me a while to figure this out). 

See description here: https://github.com/technyon/nuki_hub?tab=readme-ov-file#pairing-with-a-nuki-lock-ultra

AzonInc, 2025-02-27:
That's cool.
Could you please provide some logs for idf when compile fails?

AzonInc, 2025-05-01:
Did you set the pairing pin first?

brotaxt, 2025-05-04:
Yes, in the device config under the lock: section

brotaxt, 2025-05-13:
I've tried it once again, I keep getting the mentioned error message

```
[08:46:01][D][esp-idf:000]: E (72124) NukiBle: Error getting data from NVS
[08:46:01][D][esp-idf:000]: E (72622) NukiBle.Preferences: nvs_get_blob len fail: bleAddress NOT_FOUND
```

mhier, 2025-09-07:
I don't really understand what's going wrong here. I rebased on the latest main (see https://github.com/mhier/ESPHome_nuki_lock/tree/ultra) but it did not help. I can easily pair the same hardware when using nuki_hub. The weird thing is that it uses the same version of NukiBle, and I don't really find any obvious difference how NukiBle is used (I saw the transmit power setting was missing, but it makes no difference either).

Any hint what could be wrong here? Maybe the build system? We have to use Arduino at the moment, as with IDF  different dependency version/fork is used...

AzonInc, 2025-11-24:
I need a tester for the following config with Ultra / Go / 5th Gen Lock.

Please use this minimal config as is and set your wifi credentials as well as the pin.
```yaml
esphome:
  name: nuki-lock
  friendly_name: Nuki Lock
  on_boot:
    - lambda: |-
        esp_log_level_set("wifi", ESP_LOG_WARN);

esp32:
  variant:  ESP32S3
  framework:
    type: esp-idf
    log_level: INFO

logger:
  level: DEBUG

wifi:
  ssid: "YourSSID"
  password: "YourPassword"

psram:
  mode: octal
  speed: 80MHz

api:

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@pairing-issue
    refresh: 10s

lock:
  - platform: nuki_lock
    name: Nuki Lock
    security_pin: 123456
    pairing_as_app: true
    is_connected:
      name: "Nuki Connected"
    is_paired:
      name: "Nuki Paired"
    pairing_mode:
      name: "Nuki Pairing Mode"
    unpair:
      name: "Nuki Unpair Device"
    on_pairing_mode_on_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned on");
    on_pairing_mode_off_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned off");
    on_paired_action:
      - lambda: ESP_LOGI("nuki_lock", "Paired sucessfuly");
```

Make sure "Bluetooth pairing" is enabled for the Nuki device by enabling this setting in the official Nuki App in "Settings" > "Features & Configuration" > "Button and LED".

brotaxt, 2025-11-25:
He complains that the security PIN is allegedly incorrect, but it is set correctly in my configuration (6 digits).

```
[12:23:17.119][D][esp-idf:000]: I (56155) NukiBle: ##################### SEND AUTHORIZATION DATA (ULTRA) #########################
[12:23:17.119][D][esp-idf:000]: I (56158) NukiBle: connecting within: loopTask
[12:23:17.119][D][esp-idf:000]: I (56160) NukiBle: [Nuki ESPHome] Connected to: 54:XX:XX:76:XX:XX RSSI: -65
[12:23:17.120][D][esp-idf:000]: I (56164) NimBLE: GATT procedure initiated: write; 
[12:23:17.124][D][esp-idf:000]: I (56167) NimBLE: att_handle=26 len=2
[12:23:17.330][D][esp-idf:000]: I (56325) NimBLE: GATT procedure initiated: write; 
[12:23:17.330][D][esp-idf:000]: I (56327) NimBLE: att_handle=25 len=94
[12:23:42.334][D][esp-idf:000]: W (81377) NukiBle: Pairing timeout
[12:23:42.411][D][esp-idf:000]: I (81429) NukiBle: pairing result 2
[12:23:42.412][W][component:454]: nuki_lock.lock took a long time for an operation (31270 ms)
[12:23:42.412][W][component:457]: Components should block for at most 30 ms
[12:23:42.412][W][component:454]: improv_serial took a long time for an operation (31272 ms)
[12:23:42.412][W][component:457]: Components should block for at most 30 ms
[12:23:42.412][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[12:23:42.413][D][esp32.preferences:149]: Writing 2 items: 1 cached, 1 written, 0 failed
[12:23:42.415][D][esp-idf:000][nimble_host]: I (50734) NimBLE: GAP procedure initiated: 
[12:23:42.418][D][esp-idf:000][nimble_host]: I (50734) NimBLE: connection parameter update; conn_handle=0 itvl_min=12 itvl_max=12 latency=0 supervision_timeout=600 min_ce_len=0 max_ce_len=0
[12:23:42.429][D][esp-idf:000][nimble_host]: I (50734) NimBLE: 
[12:23:42.439][D][esp-idf:000][nimble_host]: I (50747) NimBLE: GATT procedure initiated: exchange mtu
[12:23:42.449][D][esp-idf:000][nimble_host]: E (56461) NukiBle: Error: 21 for command: 00:06
[12:23:42.514][I][nuki_lock.lock:1843][nimble_host]: Event notified 2
[12:23:42.514][W][nuki_lock.lock:1850][nimble_host]: Invalid security pin - please check (5288)
[12:23:42.514][D][esp-idf:000]: I (81528) NimBLE: GAP procedure initiated: discovery; 
[12:23:42.514][D][esp-idf:000]: I (81529) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=0 
[12:23:42.514][D][esp-idf:000]: I (81530) NimBLE: duration=forever
[12:23:42.514][D][esp-idf:000]: I (81538) NimBLE: 
[12:23:42.528][D][esp-idf:000]: E (81573) NukiBle: No credentials found - not paired yet!
[12:23:42.529][D][esp-idf:000]: I (81575) NukiBle: No nuki in pairing mode found
[12:23:42.530][D][esp-idf:000]: I (81576) NukiBle: pairing result 0
```

brotaxt, 2025-11-25:
full log. I'm using a nuki gen5 pro

```
[12:31:45.558][D][esp-idf:000]: I (564599) NukiBle: No nuki in pairing mode found
[12:31:45.560][D][esp-idf:000]: I (564601) NukiBle: pairing result 0
[12:31:46.059][D][esp-idf:000]: E (565098) NukiBle: No credentials found - not paired yet!
[12:31:46.059][D][esp-idf:000]: I (565100) NukiBle: No nuki in pairing mode found
[12:31:46.062][D][esp-idf:000]: I (565102) NukiBle: pairing result 0
[12:31:46.563][D][esp-idf:000]: E (565599) NukiBle: No credentials found - not paired yet!
[12:31:46.564][D][esp-idf:000]: I (565601) NukiBle: No nuki in pairing mode found
[12:31:46.564][D][esp-idf:000]: I (565602) NukiBle: pairing result 0
[12:31:47.074][D][esp-idf:000]: E (566100) NukiBle: No credentials found - not paired yet!
[12:31:47.074][D][esp-idf:000]: I (566102) NukiBle: No nuki in pairing mode found
[12:31:47.074][D][esp-idf:000]: I (566103) NukiBle: pairing result 0
[12:31:47.484][D][esp-idf:000][nimble_host]: I (566472) NukiBle: Found nuki ultra in pairing state:  addr: XX:d2:XX:XX:0e:XX
[12:31:47.585][D][esp-idf:000]: E (566601) NukiBle: No credentials found - not paired yet!
[12:31:47.585][D][esp-idf:000]: I (566603) NukiBle: Nuki in pairing mode found
[12:31:47.585][D][esp-idf:000]: I (566606) NukiBle: connecting within: loopTask
[12:31:47.585][D][esp-idf:000]: I (566608) NimBLE: GAP procedure initiated: connect; 
[12:31:47.585][D][esp-idf:000]: I (566610) NimBLE: peer_addr_type=0 peer_addr=
[12:31:47.585][D][esp-idf:000]: I (566612) NimBLE: XX:d2:XX:XX:0e:XX
[12:31:47.587][D][esp-idf:000]: I (566619) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
[12:31:47.588][D][esp-idf:000]: I (566630) NimBLE: 
[12:31:47.793][D][esp-idf:000]: I (566797) NukiBle: [Nuki ESPHome] Failed to connect
[12:31:47.811][D][esp-idf:000]: I (566849) NukiBle: [Nuki ESPHome] Connected to: XX:d2:XX:XX:0e:XX RSSI: -59
[12:31:47.813][D][esp-idf:000]: I (566854) NimBLE: GATT procedure initiated: write; 
[12:31:47.814][D][esp-idf:000]: I (566856) NimBLE: att_handle=26 len=2
[12:31:48.012][D][esp-idf:000]: I (567049) NukiBle: ##################### REQUEST REMOTE PUBLIC KEY #########################
[12:31:48.013][D][esp-idf:000]: I (567053) NukiBle: connecting within: loopTask
[12:31:48.020][D][esp-idf:000]: I (567055) NukiBle: [Nuki ESPHome] Connected to: XX:d2:XX:XX:0e:XX RSSI: -61
[12:31:48.022][D][esp-idf:000]: I (567063) NimBLE: GATT procedure initiated: write; 
[12:31:48.024][D][esp-idf:000]: I (567065) NimBLE: att_handle=26 len=2
[12:31:48.238][D][esp-idf:000]: I (567277) NimBLE: GATT procedure initiated: write; 
[12:31:48.239][D][esp-idf:000]: I (567279) NimBLE: att_handle=25 len=6
[12:31:50.045][D][esp-idf:000]: I (569077) NukiBle: ##################### SEND CLIENT PUBLIC KEY #########################
[12:31:50.045][D][esp-idf:000]: I (569079) NukiBle: connecting within: loopTask
[12:31:50.045][D][esp-idf:000]: I (569081) NukiBle: [Nuki ESPHome] Connected to: XX:d2:XX:XX:0e:XX RSSI: -68
[12:31:50.047][D][esp-idf:000]: I (569086) NimBLE: GATT procedure initiated: write; 
[12:31:50.051][D][esp-idf:000]: I (569089) NimBLE: att_handle=26 len=2
[12:31:50.249][D][esp-idf:000]: I (569257) NimBLE: GATT procedure initiated: write; 
[12:31:50.249][D][esp-idf:000]: I (569259) NimBLE: att_handle=25 len=36
[12:31:50.283][D][esp-idf:000]: I (569323) NukiBle: ##################### CALCULATE DH SHARED KEY s #########################
[12:31:50.351][D][esp-idf:000]: I (569348) NukiBle: ##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[12:31:50.453][D][esp-idf:000]: I (569450) NukiBle: ##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[12:31:50.463][D][esp-idf:000]: I (569503) NukiBle: ##################### SEND AUTHENTICATOR #########################
[12:31:50.464][D][esp-idf:000]: I (569505) NukiBle: connecting within: loopTask
[12:31:50.468][D][esp-idf:000]: I (569507) NukiBle: [Nuki ESPHome] Connected to: XX:d2:XX:XX:0e:XX RSSI: -64
[12:31:50.470][D][esp-idf:000]: I (569511) NimBLE: GATT procedure initiated: write; 
[12:31:50.471][D][esp-idf:000]: I (569513) NimBLE: att_handle=26 len=2
[12:31:50.658][D][esp-idf:000]: I (569677) NimBLE: GATT procedure initiated: write; 
[12:31:50.658][D][esp-idf:000]: I (569679) NimBLE: att_handle=25 len=36
[12:31:52.399][D][esp-idf:000]: I (571392) NukiBle: ##################### SEND AUTHORIZATION DATA (ULTRA) #########################
[12:31:52.399][D][esp-idf:000]: I (571395) NukiBle: connecting within: loopTask
[12:31:52.399][D][esp-idf:000]: I (571397) NukiBle: [Nuki ESPHome] Connected to: XX:d2:XX:XX:0e:XX RSSI: -64
[12:31:52.399][D][esp-idf:000]: I (571402) NimBLE: GATT procedure initiated: write; 
[12:31:52.399][D][esp-idf:000]: I (571404) NimBLE: att_handle=26 len=2
[12:31:52.502][D][esp-idf:000]: I (571537) NimBLE: GATT procedure initiated: write; 
[12:31:52.502][D][esp-idf:000]: I (571539) NimBLE: att_handle=25 len=94
[12:32:18.038][D][esp-idf:000]: W (597077) NukiBle: Pairing timeout
[12:32:18.102][D][esp-idf:000]: I (597129) NukiBle: pairing result 2
[12:32:18.103][D][esp-idf:000][nimble_host]: I (566786) NimBLE: GAP procedure initiated: 
[12:32:18.103][D][esp-idf:000][nimble_host]: I (566787) NimBLE: connection parameter update; conn_handle=0 itvl_min=12 itvl_max=12 latency=0 supervision_timeout=600 min_ce_len=0 max_ce_len=0
[12:32:18.103][D][esp-idf:000][nimble_host]: I (566787) NimBLE: 
[... cut ...]
[12:32:21.174][D][esp-idf:000]: I (600206) NukiBle: No nuki in pairing mode found
[12:32:21.174][D][esp-idf:000]: I (600207) NukiBle: pairing result 0
[12:32:21.686][D][esp-idf:000]: E (600705) NukiBle: No credentials found - not paired yet!
[12:32:21.687][D][esp-idf:000]: I (600707) NukiBle: No nuki in pairing mode found
[12:32:21.687][D][esp-idf:000]: I (600708) NukiBle: pairing result 0
[12:32:22.197][D][esp-idf:000]: E (601206) NukiBle: No credentials found - not paired yet!
[12:32:22.197][D][esp-idf:000]: I (601208) NukiBle: No nuki in pairing mode found
```

AzonInc, 2025-11-25:
What about the initialization of ESPHome?
You probably can't connect via serial because its a wall switch right?

The log output shows a 4 digit pin in the failed message. Did you trim that one?

AzonInc, 2025-11-25:
I also changed the preferences store name so the override pin will be cleared for everyone updating.
Most of the people don't use the override method anyway - so it's safe to change.

brotaxt, 2025-11-25:
This esphome init log shows Pin Status: Invalid:

```
[12:48:39.105][C][nuki_lock.lock:1765]: nuki_lock:
[12:48:39.105][C][nuki_lock.lock:1771]:   Event: Disabled
[12:48:39.105][C][nuki_lock.lock:1777]:   Pairing Identity: App
[12:48:39.105][C][nuki_lock.lock:1779]:   Pairing mode timeout: 300s
[12:48:39.107][C][nuki_lock.lock:1780]:   Configuration query interval: 3600s
[12:48:39.107][C][nuki_lock.lock:1781]:   Auth Data query interval: 3600s
[12:48:39.115][C][nuki_lock.lock:1782]:   BLE general timeout: 3s
[12:48:39.115][C][nuki_lock.lock:1783]:   BLE command timeout: 3s
[12:48:39.124][C][nuki_lock.lock:1787]:   Pin Status: Invalid
[12:48:39.125][C][nuki_lock.lock:1789]: nuki_lock.lockNuki Lock 'Nuki Lock'
[12:48:39.134][C][nuki_lock.lock:018]: nuki_lock.lockIs Connected 'Nuki Connected'
[12:48:39.135][C][nuki_lock.lock:021]: nuki_lock.lock  Device Class: 'connectivity'
[12:48:39.144][C][nuki_lock.lock:018]: nuki_lock.lockIs Paired 'Nuki Paired'
[12:48:39.154][C][nuki_lock.lock:021]: nuki_lock.lock  Device Class: 'connectivity'
[12:48:39.209][C][nuki_lock.lock:092]: nuki_lock.lockPairing Mode 'Nuki Pairing Mode'
[12:48:39.209]nuki_lock.lock  Restore Mode: always OFF
[12:48:39.209][C][nuki_lock.lock:100]: nuki_lock.lock  Icon: 'mdi:bluetooth'
```

brotaxt, 2025-11-25:
Same result

```
[13:02:37.891][D][nuki_lock.lock:1521]: Using security pin from yaml config to pair.
[13:02:37.893][D][esp-idf:000]: E (48255) NukiBle: No credentials found - not paired yet!
[13:02:37.895][D][esp-idf:000]: I (48258) NukiBle: No nuki in pairing mode found
[13:02:37.897][D][esp-idf:000]: I (48260) NukiBle: pairing result 0
[13:02:38.395][D][nuki_lock.lock:1521]: Using security pin from yaml config to pair.
[13:02:38.397][D][esp-idf:000]: E (48759) NukiBle: No credentials found - not paired yet!
[13:02:38.399][D][esp-idf:000]: I (48762) NukiBle: No nuki in pairing mode found
[13:02:38.401][D][esp-idf:000]: I (48764) NukiBle: pairing result 0
[13:02:38.901][D][nuki_lock.lock:1521]: Using security pin from yaml config to pair.
[13:02:38.902][D][esp-idf:000]: E (49261) NukiBle: No credentials found - not paired yet!
[13:02:38.902][D][esp-idf:000]: I (49264) NukiBle: No nuki in pairing mode found
[13:02:38.905][D][esp-idf:000]: I (49265) NukiBle: pairing result 0
[13:02:38.997][D][esp-idf:000][nimble_host]: I (49350) NukiBle: Found nuki ultra in pairing state:  addr: X:X:X:X:X:X
[13:02:39.207][D][esp-idf:000][nimble_host]: I (49564) NukiBle: Found nuki ultra in pairing state:  addr: X:X:X:X:X:X
[13:02:39.408][D][nuki_lock.lock:1521]: Using security pin from yaml config to pair.
[13:02:39.408][D][esp-idf:000]: E (49763) NukiBle: No credentials found - not paired yet!
[13:02:39.408][D][esp-idf:000]: I (49765) NukiBle: Nuki in pairing mode found
[13:02:39.408][D][esp-idf:000]: I (49767) NukiBle: connecting within: loopTask
[13:02:39.408][D][esp-idf:000]: I (49769) NimBLE: GAP procedure initiated: connect; 
[13:02:39.413][D][esp-idf:000]: I (49771) NimBLE: peer_addr_type=0 peer_addr=
[13:02:39.424][D][esp-idf:000]: I (49777) NimBLE: X:X:X:X:X:X
[13:02:39.425][D][esp-idf:000]: I (49787) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
[13:02:39.433][D][esp-idf:000]: I (49789) NimBLE: 
[13:02:40.146][D][esp-idf:000]: I (50506) NukiBle: [Nuki ESPHome] Failed to connect
[13:02:40.196][D][esp-idf:000]: I (50558) NimBLE: GAP procedure initiated: connect; 
[13:02:40.197][D][esp-idf:000]: I (50560) NimBLE: peer_addr_type=0 peer_addr=
[13:02:40.199][D][esp-idf:000]: I (50561) NimBLE: X:X:X:X:X:X
[13:02:40.204][D][esp-idf:000]: I (50562) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
[13:02:40.212][D][esp-idf:000]: I (50566) NimBLE: 
[13:02:40.543][D][esp-idf:000]: I (50904) NukiBle: [Nuki ESPHome] Failed to connect
[13:02:40.634][D][esp-idf:000]: I (50956) NukiBle: [Nuki ESPHome] Connected to: X:X:X:X:X:X RSSI: -60
[13:02:40.634][D][esp-idf:000]: I (50961) NimBLE: GATT procedure initiated: discover service by uuid; uuid=
[13:02:40.634][D][esp-idf:000]: I (50963) NimBLE: a92ee300-5501-11e4-916c-0800200c9a66
[13:02:40.634][D][esp-idf:000]: I (50964) NimBLE: 
[13:02:40.837][D][esp-idf:000]: I (51137) NimBLE: GATT procedure initiated: discover characteristics by uuid; 
[13:02:40.838][D][esp-idf:000]: I (51139) NimBLE: start_handle=23 end_handle=26 uuid=
[13:02:40.838][D][esp-idf:000]: I (51140) NimBLE: a92ee301-5501-11e4-916c-0800200c9a66
[13:02:40.839][D][esp-idf:000]: I (51141) NimBLE: 
[13:02:40.941][D][esp-idf:000]: I (51297) NimBLE: GATT procedure initiated: discover all descriptors; 
[13:02:40.942][D][esp-idf:000]: I (51299) NimBLE: chr_val_handle=25 end_handle=26
[13:02:41.043][D][esp-idf:000]: I (51372) NimBLE: GATT procedure initiated: write; 
[13:02:41.043][D][esp-idf:000]: I (51374) NimBLE: att_handle=26 len=2
[13:02:41.160][D][esp-idf:000]: I (51520) NukiBle: ##################### REQUEST REMOTE PUBLIC KEY #########################
[13:02:41.161][D][esp-idf:000]: I (51523) NukiBle: connecting within: loopTask
[13:02:41.165][D][esp-idf:000]: I (51525) NukiBle: [Nuki ESPHome] Connected to: X:X:X:X:X:X RSSI: -63
[13:02:41.167][D][esp-idf:000]: I (51530) NimBLE: GATT procedure initiated: write; 
[13:02:41.169][D][esp-idf:000]: I (51532) NimBLE: att_handle=26 len=2
[13:02:41.351][D][esp-idf:000]: I (51682) NimBLE: GATT procedure initiated: write; 
[13:02:41.351][D][esp-idf:000]: I (51684) NimBLE: att_handle=25 len=6
[13:02:43.093][D][esp-idf:000]: I (53447) NukiBle: ##################### SEND CLIENT PUBLIC KEY #########################
[13:02:43.093][D][esp-idf:000]: I (53449) NukiBle: connecting within: loopTask
[13:02:43.093][D][esp-idf:000]: I (53451) NukiBle: [Nuki ESPHome] Connected to: X:X:X:X:X:X RSSI: -60
[13:02:43.093][D][esp-idf:000]: I (53455) NimBLE: GATT procedure initiated: write; 
[13:02:43.094][D][esp-idf:000]: I (53457) NimBLE: att_handle=26 len=2
[13:02:43.296][D][esp-idf:000]: I (53632) NimBLE: GATT procedure initiated: write; 
[13:02:43.296][D][esp-idf:000]: I (53634) NimBLE: att_handle=25 len=36
[13:02:43.336][D][esp-idf:000]: I (53697) NukiBle: ##################### CALCULATE DH SHARED KEY s #########################
[13:02:43.398][D][esp-idf:000]: I (53724) NukiBle: ##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[13:02:43.501][D][esp-idf:000]: I (53827) NukiBle: ##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[13:02:43.518][D][esp-idf:000]: I (53880) NukiBle: ##################### SEND AUTHENTICATOR #########################
[13:02:43.519][D][esp-idf:000]: I (53882) NukiBle: connecting within: loopTask
[... cut ...]
[13:03:45.258][D][esp-idf:000]: E (115616) NukiBle: No credentials found - not paired yet!
[13:03:45.258][D][esp-idf:000]: I (115617) NukiBle: No nuki in pairing mode found
[13:03:45.258][D][esp-idf:000]: I (115619) NukiBle: pairing result 0
[13:03:45.761][D][nuki_lock.lock:1521]: Using security pin from yaml config to pair.
[13:03:45.761][D][esp-idf:000]: E (116116) NukiBle: No credentials found - not paired yet!
[13:03:45.761][D][esp-idf:000]: I (116118) NukiBle: No nuki in pairing mode found
[13:03:45.761][D][esp-idf:000]: I (116119) NukiBle: pairing result 0
```

AzonInc, 2025-11-25:
The log now shows a 5 digit pin, is any number of that correct?

NukiBle: ##################### AUTH 10416 (ULTRA) ####################

AzonInc, 2025-11-25:
after AUTH in the message I sent

brotaxt, 2025-11-25:
There is nothing. Ive posted the full log

```
[13:02:45.446][D][esp-idf:000]: I (55754) NukiBle: ##################### AUTH 10416 (ULTRA) ####################
[13:02:45.446][D][esp-idf:000]: I (55756) NukiBle: connecting within: loopTask
[13:02:45.446][D][esp-idf:000]: I (55758) NukiBle: [Nuki ESPHome] Connected to: X:X:X:X:X:X RSSI: -62
[13:02:45.446][D][esp-idf:000]: I (55764) NimBLE: GATT procedure initiated: write; 
[13:02:45.446][D][esp-idf:000]: I (55766) NimBLE: att_handle=26 len=2
[13:02:45.549][D][esp-idf:000]: I (55897) NimBLE: GATT procedure initiated: write; 
[13:02:45.549][D][esp-idf:000]: I (55899) NimBLE: att_handle=25 len=94
[13:03:11.172][D][esp-idf:000]: W (81533) NukiBle: Pairing timeout
```

AzonInc, 2025-11-25:
10416, but as you don't recognize the number I assume it's not even close to your pin.

AzonInc, 2025-11-25:
**NVS pin value (ultra/go/gen5):**

never shows 0 for you, right?
It shows a 4 digit pin?

As well as
**NVS pin value (gen 1-4):**

AzonInc, 2025-11-25:
Which Lock do you use and may I ask for the pin? You can also send me a message on Discord if thats better. Maybe the pin is special as it worked for someone else.

brotaxt, 2025-11-25:
Yeah my pin number started with a zero. I changed the pin to a new value, did another factory reset and reflashed, its working now  :) The previous pin was working in the app altough

AzonInc, 2025-11-25:
I think you might be able to use padding zero Pins if you remove the zeros.

001234 -> 1234