uriyacovy/ESPHome_nuki_lock #70: Pairing process cause a reboot

mr2c12, 2025-01-17:
When the pairing process start (esp32 in pairing mode, than 5-seconds press of button on Nuki v2) my Wroom esp32 board reboot.

This is the log collected via usb:

> [22:22:46][I][nuki_lock.lock:1181]: Waiting for Nuki to enter pairing mode...
> [22:22:46][ 35097][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:46][ 35098][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:47][ 35599][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:47][ 35599][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:47][ 36095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:47][ 36095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:48][ 36595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:48][ 36595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:48][ 37095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:48][ 37095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:49][ 37595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:49][ 37595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:49][ 38095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:49][ 38095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:50][ 38594][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:50][ 38594][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:50][ 39094][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:50][ 39094][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:51][ 39594][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:51][ 39594][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:51][ 40094][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:51][ 40094][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:52][ 40594][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:52][ 40594][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:52][ 41094][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:52][ 41094][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:53][ 41594][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:53][ 41594][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:53][ 42094][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:53][ 42094][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:54][ 42600][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:54][ 42600][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:54][ 43095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:54][ 43095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:55][ 43595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:55][ 43595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:55][ 44095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:55][ 44095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:56][ 44595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:56][ 44595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:56][ 45095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:56][ 45095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:57][ 45595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:57][ 45595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:57][ 46095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:57][ 46095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:58][ 46595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:58][ 46595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:58][ 47095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:58][ 47095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:59][ 47595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:59][ 47595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:22:59][ 48095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:22:59][ 48095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:23:00][ 48595][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:23:00][ 48595][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:23:00][ 49095][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bleAddress NOT_FOUND
> [22:23:00][ 49095][E][NukiBle.cpp:939] retrieveCredentials(): Error getting data from NVS
> [22:23:01]
> [22:23:01]abort() was called at PC 0x400fa304 on core 1
> [22:23:01]
> [22:23:01]
> [22:23:01]Backtrace:0x4008397d:0x3ffcd6700x40093351:0x3ffcd690 0x40098ef5:0x3ffcd6b0 0x400fa304:0x3ffcd730 0x400fa259:0x3ffcd750 0x400fa162:0x3ffcd7b0 0x400fa17a:0x3ffcd7d0 0x400fa183:0x3ffcd7f0 0x40130df9:0x3ffcd810 0x40130de5:0x3ffcd830 0x400f3e25:0x3ffcd850 0x400e0d19:0x3ffcd8a0 0x400e0d66:0x3ffcd900 0x401a7a31:0x3ffcd920 0x400eb037:0x3ffcd940 0x400e89d9:0x3ffcd990 0x400eb82e:0x3ffcd9c0 0x400ff275:0x3ffcd9e0 
> WARNING Found stack trace! Trying to decode it
> WARNING Decoded 0x4008397d: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
[... 15 lines cut ...]
> WARNING Decoded 0x400eb037: std::function<void ()>::operator()() const at /home/user/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
>  (inlined by) esphome::Scheduler::call() at /home/user/Documenti/VSCode workspace/Esphome/.esphome/build/rscurtains/src/esphome/core/scheduler.cpp:234
> WARNING Decoded 0x400e89d9: esphome::Application::loop() at /home/user/Documenti/VSCode workspace/Esphome/.esphome/build/rscurtains/src/esphome/core/application.cpp:69
> WARNING Decoded 0x400eb82e: loop() at /home/user/Documenti/VSCode workspace/Esphome/.esphome/build/rscurtains/rscurtains_esp32.yaml:220
> WARNING Decoded 0x400ff275: loopTask(void*) at /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
> [22:23:02]
> [22:23:02]
> [22:23:02]
> [22:23:02]
> [22:23:02]ELF file SHA256: 0000000000000000
> [22:23:02]
> [22:23:02]Rebooting...
> [22:23:02]ets Jun  8 2016 00:22:57
> [22:23:02]
> [22:23:02]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> [22:23:02]configsip: 0, SPIWP:0xee
> [22:23:02]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> [22:23:02]mode:DIO, clock div:2
> [22:23:02]load:0x3fff0030,len:1184
> [22:23:02]load:0x40078000,len:13132
> [22:23:02]load:0x40080400,len:3036
> [22:23:02]entry 0x400805e4
> [22:23:02][I][logger:171]: Log initialized

Since I have other sensors already defined in the same yaml, can it be a ram problem?

AzonInc, 2025-01-17:
Most of the time it's a memory issue.
Which other components are you using?
Also encryption is taking quite some memory as well. Seems like this error is originating from encryption.

mr2c12, 2025-01-18:
Once the encryption has been disabled, I have been able to successfully paired.

Now it seems to work, but I see a lot of these lines:

> ][E][nuki_lock.lock:431]: requestAdvancedConfig has resulted in failed (2)

AzonInc, 2025-01-18:
That's more or less normal. 
The lock does not always immediately respond to requests, so it retries until its successful.

mr2c12, 2025-01-18:
Thank you. Please consider this issue fixed, the.

closed 2025-01-18 by mr2c12