uriyacovy/ESPHome_nuki_lock #75: "Error getting data from NVS"

kooskaspers, 2025-02-17:
I'm using this POE esp32: https://docs.m5stack.com/en/unit/poesp32
Flashed using the following variables:

```
external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock

esphome:
  name: nukilock
  friendly_name: NukiLock

esp32:
  board: m5stack-core-esp32

# Enable logging
logger:

ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  power_pin: GPIO5  
  phy_addr: 1

# Enable Home Assistant API
api:

lock:
  # Required
  - platform: nuki_lock
    name: Nuki Lock
  # Optional: Settings
    pairing_mode_timeout: 300s
    event: "nuki"
    security_pin: 5452
  # Optional: Advanced Settings
    alternative_connect_mode: true
    pairing_as_app: false
  # Optional: Binary Sensors
    is_connected:
      name: "Nuki Connected"
    is_paired:
      name: "Nuki Paired"
    battery_critical:
      name: "Nuki Battery Critical"
    door_sensor:
      name: "Nuki Door Sensor"
  # Optional: Sensors
    battery_level:
      name: "Nuki Battery Level"
    bt_signal_strength:
      name: "Bluetooth Signal Strength"
  # Optional: Text Sensors
    door_sensor_state:
      name: "Nuki Door Sensor: State"
    last_unlock_user:
      name: "Nuki Last Unlock User"
    last_lock_action:
      name: "Nuki Last Lock Action"
    last_lock_action_trigger:
      name: "Nuki Last Lock Action Trigger"
  # Optional: Switches
    pairing_mode:
      name: "Nuki Pairing Mode"
    auto_unlatch:
      name: "Nuki Auto unlatch"
    button_enabled:
      name: "Nuki Button: Locking operations"
    led_enabled:
      name: "Nuki LED Signal"
  # Optional: Number Inputs
    led_brightness:
      name: "Nuki LED Brightness"
    timezone_offset:
      name: "Nuki Timezone: Offset"
  # Optional: Switches
    night_mode_enabled:
      name: "Nuki Night Mode"
    night_mode_auto_lock_enabled:
      name: "Nuki Night Mode: Auto Lock"
    night_mode_auto_unlock_disabled:
      name: "Nuki Night Mode: Reject Auto Unlock"
    night_mode_immediate_lock_on_start_enabled:
      name: "Nuki Night Mode: Lock at Start Time"
    auto_lock_enabled:
      name: "Nuki Auto Lock"
    auto_unlock_disabled:
      name: "Nuki Auto Unlock: Disable"
    immediate_auto_lock_enabled:
      name: "Nuki Auto Lock: Immediately"
    auto_update_enabled:
      name: "Nuki Automatic Updates"
    single_lock_enabled:
      name: "Nuki Single Lock"
    dst_mode_enabled:
      name: "Nuki Daylight Saving Time"
  # Optional: Select Inputs
    single_buton_press_action:
      name: "Nuki Button: Single Press Action"
    double_buton_press_action:
      name: "Nuki Button: Double Press Action"
    fob_action_1:
      name: "Nuki Fob: Action 1"
    fob_action_2:
      name: "Nuki Fob: Action 2"
    fob_action_3:
      name: "Nuki Fob: Action 3"
    timezone:
      name: "Nuki Timezone"
    advertising_mode:
      name: "Nuki Advertising Mode"
  # Optional: Buttons
    unpair:
      name: "Nuki Unpair Device"
  # Optional: Callbacks
    on_pairing_mode_on_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned on");
    on_pairing_mode_off_action:
      - lambda: ESP_LOGI("nuki_lock", "Pairing mode turned off");
    on_paired_action:
      - lambda: ESP_LOGI("nuki_lock", "Paired sucessfuly");
```

I'm able to add the device to home-assistant. But pairing failes. The only error i'm getting in the logs of esphome is:
`[D][esp-idf:000]: Error getting data from NVS`

Any idea what could be the problem?

AzonInc, 2025-02-17:
What's the distance to your lock?
Could you provide the fill logs please? Thx

kooskaspers, 2025-02-17:
Distance is aprox 3-4 meter.

Logs:
```
esphome logs M5Stack_poesp32kk.yaml
INFO ESPHome 2024.12.4
INFO Reading configuration M5Stack_poesp32kk.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Consider enabling PSRAM support if it's available for the NimBLE Stack.
INFO Starting log output from nukilock.local using esphome API
INFO Successfully connected to nukilock @ 192.168.1.159 in 0.889s
INFO Successful handshake with nukilock @ 192.168.1.159 in 0.019s
[22:53:38][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 17 2025, 22:28:51
[22:53:38][C][logger:185]: Logger:
[22:53:38][C][logger:186]:   Level: DEBUG
[22:53:38][C][logger:188]:   Log Baud Rate: 115200
[22:53:38][C][logger:189]:   Hardware UART: UART0
[22:53:38][C][nuki_lock.lock:1318]: NUKI LOCK:
[22:53:38][C][nuki_lock.lock:1320]: nuki_lock.lockNuki Lock 'Nuki Lock'
[22:53:38][C][nuki_lock.lock:1322]: nuki_lock.lockIs Connected 'Nuki Connected'
[22:53:38][C][nuki_lock.lock:1322]: nuki_lock.lock  Device Class: 'connectivity'
[22:53:38][C][nuki_lock.lock:1323]: nuki_lock.lockIs Paired 'Nuki Paired'
[22:53:38][C][nuki_lock.lock:1323]: nuki_lock.lock  Device Class: 'connectivity'
[22:53:38][C][nuki_lock.lock:1324]: nuki_lock.lockBattery Critical 'Nuki Battery Critical'
[22:53:38][C][nuki_lock.lock:1324]: nuki_lock.lock  Device Class: 'battery'
[22:53:38][C][nuki_lock.lock:1325]: nuki_lock.lockDoor Sensor 'Nuki Door Sensor'
[22:53:38][C][nuki_lock.lock:1325]: nuki_lock.lock  Device Class: 'door'
[22:53:38][C][nuki_lock.lock:1328]: nuki_lock.lockDoor Sensor State 'Nuki Door Sensor: State'
[22:53:38][C][nuki_lock.lock:1328]: nuki_lock.lock  Icon: 'mdi:door-open'
[22:53:38][C][nuki_lock.lock:1329]: nuki_lock.lockLast Unlock User 'Nuki Last Unlock User'
[22:53:38][C][nuki_lock.lock:1329]: nuki_lock.lock  Icon: 'mdi:account-clock'
[22:53:38][C][nuki_lock.lock:1330]: nuki_lock.lockLast Lock Action 'Nuki Last Lock Action'
[22:53:38][C][nuki_lock.lock:1330]: nuki_lock.lock  Icon: 'mdi:account-clock'
[22:53:38][C][nuki_lock.lock:1331]: nuki_lock.lockLast Lock Action Trigger 'Nuki Last Lock Action Trigger'
[22:53:38][C][nuki_lock.lock:1331]: nuki_lock.lock  Icon: 'mdi:account-clock'
[22:53:38][C][nuki_lock.lock:1334]: nuki_lock.lockBattery Level 'Nuki Battery Level'
[22:53:38][C][nuki_lock.lock:1334]: nuki_lock.lock  Device Class: 'battery'
[22:53:38][C][nuki_lock.lock:1334]: nuki_lock.lock  State Class: ''
[22:53:38][C][nuki_lock.lock:1334]: nuki_lock.lock  Unit of Measurement: '%'
[22:53:38][C][nuki_lock.lock:1334]: nuki_lock.lock  Accuracy Decimals: 0
[22:53:38][C][nuki_lock.lock:1334]: nuki_lock.lock  Icon: 'mdi:battery-50'
[22:53:38][C][nuki_lock.lock:1335]: nuki_lock.lockBluetooth Signal 'Bluetooth Signal Strength'
[22:53:38][C][nuki_lock.lock:1335]: nuki_lock.lock  Device Class: 'signal_strength'
[22:53:38][C][nuki_lock.lock:1335]: nuki_lock.lock  State Class: ''
[22:53:38][C][nuki_lock.lock:1335]: nuki_lock.lock  Unit of Measurement: 'dBm'
[22:53:38][C][nuki_lock.lock:1335]: nuki_lock.lock  Accuracy Decimals: 0
[22:53:38][C][nuki_lock.lock:1335]: nuki_lock.lock  Icon: 'mdi:bluetooth-audio'
[22:53:38][C][nuki_lock.lock:068]: nuki_lock.lockPairing Mode 'Nuki Pairing Mode'
[22:53:38][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:bluetooth'
[22:53:38][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[22:53:38][C][nuki_lock.lock:068]: nuki_lock.lockAuto Unlatch Enabled 'Nuki Auto unlatch'
[22:53:38][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:auto-upload'
[22:53:38][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[22:53:38][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[22:53:38][C][nuki_lock.lock:068]: nuki_lock.lockButton Enabled 'Nuki Button: Locking operations'
[22:53:38][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:radiobox-marked'
[22:53:38][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[22:53:38][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[22:53:38][C][nuki_lock.lock:068]: nuki_lock.lockLED Enabled 'Nuki LED Signal'
[22:53:38][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:led-on'
[22:53:38][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[22:53:38][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[22:53:38][C][nuki_lock.lock:068]: nuki_lock.lockNight Mode Enabled 'Nuki Night Mode'
[22:53:38][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:shield-moon'
[22:53:38][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[22:53:38][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[22:53:38][C][nuki_lock.lock:068]: nuki_lock.lockNight Mode Auto Lock 'Nuki Night Mode: Auto Lock'
[22:53:38][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:lock-clock'
[... 50 lines cut ...]
[22:53:39][C][nuki_lock.lock:1364]: nuki_lock.lockAdvertising Mode 'Nuki Advertising Mode'
[22:53:39][C][nuki_lock.lock:1364]: nuki_lock.lock  Icon: 'mdi:timer-cog'
[22:53:39][C][ethernet:326]: Ethernet:
[22:53:39][C][ethernet:515]:   IP Address: 192.168.1.159
[22:53:39][C][ethernet:516]:   Hostname: 'nukilock'
[22:53:39][C][ethernet:517]:   Subnet: 255.255.255.0
[22:53:39][C][ethernet:518]:   Gateway: 192.168.1.1
[22:53:39][C][ethernet:523]:   DNS1: 192.168.1.1
[... cut ...]
[22:53:39][C][ethernet:345]:   Power Pin: 5
[22:53:39][C][ethernet:347]:   MDC Pin: 23
[22:53:39][C][ethernet:348]:   MDIO Pin: 18
[22:53:39][C][ethernet:349]:   PHY addr: 1
[22:53:39][C][ethernet:351]:   Type: IP101
[22:53:39][C][mdns:116]: mDNS:
[22:53:39][C][mdns:117]:   Hostname: nukilock
[22:53:39][C][api:140]: API Server:
[22:53:39][C][api:141]:   Address: nukilock.local:6053
[22:53:39][C][api:145]:   Using noise encryption: NO
```

I already erased the device redid the flashing. But to no avail :(.

AzonInc, 2025-02-17:
Okay, uhm try pairing from 1m distance as well.
How did you try to pair, just to reproduce it.

kooskaspers, 2025-02-18:
Thank you for the suggestion. I got myself a longer UTP cable and moved the device very close to the lock.
Same problem:
```

INFO Trying to connect to nukilock.local in the background
INFO Successfully connected to nukilock @ 192.168.1.159 in 0.616s
INFO Successful handshake with nukilock @ 192.168.1.159 in 0.028s
[20:08:39][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 17 2025, 22:28:51
[20:08:39][C][logger:185]: Logger:
[20:08:39][C][logger:186]:   Level: DEBUG
[20:08:39][C][logger:188]:   Log Baud Rate: 115200
[20:08:39][C][logger:189]:   Hardware UART: UART0
[20:08:39][C][nuki_lock.lock:1318]: NUKI LOCK:
[20:08:39][C][nuki_lock.lock:1320]: nuki_lock.lockNuki Lock 'Nuki Lock'
[20:08:39][C][nuki_lock.lock:1322]: nuki_lock.lockIs Connected 'Nuki Connected'
[20:08:39][C][nuki_lock.lock:1322]: nuki_lock.lock  Device Class: 'connectivity'
[20:08:39][C][nuki_lock.lock:1323]: nuki_lock.lockIs Paired 'Nuki Paired'
[20:08:39][C][nuki_lock.lock:1323]: nuki_lock.lock  Device Class: 'connectivity'
[20:08:39][C][nuki_lock.lock:1324]: nuki_lock.lockBattery Critical 'Nuki Battery Critical'
[20:08:39][C][nuki_lock.lock:1324]: nuki_lock.lock  Device Class: 'battery'
[20:08:39][C][nuki_lock.lock:1325]: nuki_lock.lockDoor Sensor 'Nuki Door Sensor'
[20:08:39][C][nuki_lock.lock:1325]: nuki_lock.lock  Device Class: 'door'
[20:08:39][C][nuki_lock.lock:1328]: nuki_lock.lockDoor Sensor State 'Nuki Door Sensor: State'
[20:08:39][C][nuki_lock.lock:1328]: nuki_lock.lock  Icon: 'mdi:door-open'
[20:08:39][C][nuki_lock.lock:1329]: nuki_lock.lockLast Unlock User 'Nuki Last Unlock User'
[20:08:39][C][nuki_lock.lock:1329]: nuki_lock.lock  Icon: 'mdi:account-clock'
[20:08:39][C][nuki_lock.lock:1330]: nuki_lock.lockLast Lock Action 'Nuki Last Lock Action'
[20:08:39][C][nuki_lock.lock:1330]: nuki_lock.lock  Icon: 'mdi:account-clock'
[20:08:39][C][nuki_lock.lock:1331]: nuki_lock.lockLast Lock Action Trigger 'Nuki Last Lock Action Trigger'
[20:08:39][C][nuki_lock.lock:1331]: nuki_lock.lock  Icon: 'mdi:account-clock'
[20:08:39][C][nuki_lock.lock:1334]: nuki_lock.lockBattery Level 'Nuki Battery Level'
[20:08:39][C][nuki_lock.lock:1334]: nuki_lock.lock  Device Class: 'battery'
[20:08:39][C][nuki_lock.lock:1334]: nuki_lock.lock  State Class: ''
[20:08:39][C][nuki_lock.lock:1334]: nuki_lock.lock  Unit of Measurement: '%'
[20:08:39][C][nuki_lock.lock:1334]: nuki_lock.lock  Accuracy Decimals: 0
[20:08:39][C][nuki_lock.lock:1334]: nuki_lock.lock  Icon: 'mdi:battery-50'
[20:08:39][C][nuki_lock.lock:1335]: nuki_lock.lockBluetooth Signal 'Bluetooth Signal Strength'
[20:08:39][C][nuki_lock.lock:1335]: nuki_lock.lock  Device Class: 'signal_strength'
[20:08:39][C][nuki_lock.lock:1335]: nuki_lock.lock  State Class: ''
[20:08:39][C][nuki_lock.lock:1335]: nuki_lock.lock  Unit of Measurement: 'dBm'
[20:08:39][C][nuki_lock.lock:1335]: nuki_lock.lock  Accuracy Decimals: 0
[20:08:39][C][nuki_lock.lock:1335]: nuki_lock.lock  Icon: 'mdi:bluetooth-audio'
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockPairing Mode 'Nuki Pairing Mode'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:bluetooth'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockAuto Unlatch Enabled 'Nuki Auto unlatch'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:auto-upload'
[20:08:39][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockButton Enabled 'Nuki Button: Locking operations'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:radiobox-marked'
[20:08:39][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockLED Enabled 'Nuki LED Signal'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:led-on'
[20:08:39][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockNight Mode Enabled 'Nuki Night Mode'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:shield-moon'
[20:08:39][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockNight Mode Auto Lock 'Nuki Night Mode: Auto Lock'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:lock-clock'
[20:08:39][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockNight Mode Auto Unlock Disabled 'Nuki Night Mode: Reject Auto Unlock'
[20:08:39][C][nuki_lock.lock:070]: nuki_lock.lock  Icon: 'mdi:lock-remove'
[20:08:39][C][nuki_lock.lock:079]: nuki_lock.lock  Device Class: 'switch'
[20:08:39][C][nuki_lock.lock:091]: nuki_lock.lock  Restore Mode: always OFF
[20:08:39][C][nuki_lock.lock:068]: nuki_lock.lockNight Mode Immediate Lock On Start 'Nuki Night Mode: Lock at Start Time'
[... 57 lines cut ...]
[20:08:40][C][ethernet:348]:   MDIO Pin: 18
[20:08:40][C][ethernet:349]:   PHY addr: 1
[20:08:40][C][ethernet:351]:   Type: IP101
[20:08:40][C][mdns:116]: mDNS:
[20:08:40][C][mdns:117]:   Hostname: nukilock
[20:08:40][C][api:140]: API Server:
[... cut ...]
[20:09:30][D][esp-idf:000]: Error getting data from NVS
[20:09:31][D][esp-idf:000]: Error getting data from NVS
[20:09:31][D][esp-idf:000]: Error getting data from NVS
[20:09:32][D][esp-idf:000]: Error getting data from NVS
[20:09:32][D][esp-idf:000]: Error getting data from NVS
[20:09:33][D][esp-idf:000]: Error getting data from NVS
[20:09:33][D][esp-idf:000]: Error getting data from NVS
[20:09:34][D][esp-idf:000]: Error getting data from NVS
[20:09:34][D][esp-idf:000]: Error getting data from NVS
```

AzonInc, 2025-02-18:
Another question: Which Nuki Lock are you trying to pair? Gen 1-4 or Ultra?

kooskaspers, 2025-02-19:
@AzonInc, I discovered the problem. I was able to succesfully pair the device after moving it very close (< 1 meter) to the lock (gen 2 pro). The problem is there is no antenna on the PCB for the bluetooth reception on this specific esp32 ([M5Stack PoESP32](https://docs.m5stack.com/en/unit/poesp32). There is an IPEX connector though,  so I've ordered an external antenna and I'm sure this will solve the issue. Thanks for helping out!

closed 2025-02-19 by kooskaspers