uriyacovy/ESPHome_nuki_lock #76: ESP locks up after several hours

s00500, 2025-04-13:
after an average of about 48h my esp usually becomes unresponsive until I reset it again ... I though for a long while that this might be related to the power supply but now I have a really stable one and it still happens...

Anyone else experiencing issues like these ?

AzonInc, 2025-05-05:
Did you try disableing the events by setting event: none ?
That might free up some resources and disables code for gathering event logs as well as the Home Assistant API log streaming.

s00500, 2025-05-05:
Hm yes, already running like that:

````
external_components:
  - source:
      type: git
      url: https://github.com/uriyacovy/ESPHome_nuki_lock
      ref: improve-memory-footprint
    components: [nuki_lock]

esp32:
  board: "esp32dev" # Or whatever other board you're using
  framework:
    type: arduino
    version: 2.0.16
    platform_version: 6.7.0

lock:
  # Required
  - platform: nuki_lock
    name: Main Lock
    # Optional: Settings
    pairing_mode_timeout: 300s
    event: "none"
    # Optional: Binary Sensors
````

AzonInc, 2025-05-06:
Could you please try using the IDF version: #72

flosch-dev, 2025-05-07:
I having the same issue... I will try IDF from here  https://github.com/uriyacovy/ESPHome_nuki_lock/issues/72 and I can report if problem will disappear

AzonInc, 2025-05-08:
Which ESP are y'all using?

s00500, 2025-05-11:
Ah ok, ran into a different issue, reported on other tickets, thanks for the comments!

I am using a normal ESP32 on a Wemos D1 Mini Style DevBoard

manonfgoo, 2025-05-26:
I have the same issue, logs from the serial console show the following:
`
[D][nuki_lock.lock:1110]: Update present, getting auth data...
[E][nuki_lock.lock:708]: retrieveAuthorizationEntries has resulted in failed (2)
[D][nuki_lock.lock:1110]: Update present, getting auth data...
[137905102][E][NukiBle.cpp:1596] handleReturnMessage(): Error: 21 for command: 00:09
[D][nuki_lock.lock:678]: retrieveAuthorizationEntries has resulted in success (1)
[W][nuki_lock.lock:704]: No auth entries! Did you set the security pin?
[D][time:047]: Synchronized time: 2025-05-26 16:04:42

[....]

[D][nuki_lock.lock:1110]: Update present, getting auth data...
[E][nuki_lock.lock:708]: retrieveAuthorizationEntries has resulted in failed (2)
[D][nuki_lock.lock:1110]: Update present, getting auth data...
[139214903][E][NukiBle.cpp:1596] handleReturnMessage(): Error: 21 for command: 00:09
[D][esp-idf:000][nimble_host]: K_ERROR_BAD_PIN
[D][nuki_lock.lock:1116]: Update present, getting event logs...
[E][nuki_lock.lock:745]: retrieveLogEntries has resulted in failed (2)

[....]

[D][nuki_lock.lock:1110]: Update present, getting auth data...
E (139623096) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (139623096) task_wdt:  - loopTask (CPU 1)
E (139623096) task_wdt: Tasks currently running:
E (139623096) task_wdt: CPU 0: IDLE0
`

s00500, 2025-05-26:
Update: first time the esp died today since I flashed it on the idf version, gotta say there are some other devices in my home that have needed a reset for no reason today as well... so might also be related to my network 😅

manonfgoo, 2025-06-13:
For me problems went away since adding CONFIG_ESP_PANIC_HANDLER_IRAM: "y" to sdk config

`esp32:
  board: m5stack-core2
  framework:
    type: esp-idf
    sdkconfig_options:
       CONFIG_ESP_PANIC_HANDLER_IRAM: "y" 
`

AzonInc, 2025-11-24:
Please try this branch (requires ESPHome 2025.11) if you are still experiencing stability issues:

```yaml
external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@pairing-issue
```

Do not forget to clean the build files. Maybe that solves the issue.

AzonInc, 2025-11-25:
Okay perfect, I will close this for now. Feel free to open a new issue if it happens again.