uriyacovy/ESPHome_nuki_lock #77: Using `lock.LOCK_SCHEMA` is deprecated

moon4-hacs, 2025-05-22:
WARNING Using `lock.LOCK_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `lock.lock_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: nuki_lock.

AzonInc, 2025-05-22:
When did you do the last ESPHome update (before 2025.5)?
Because we didn't change anything in the last 2 months.

moon4-hacs, 2025-05-22:
The last Update was 2025.4.2

AzonInc, 2025-05-23:
Sounds like an ESPHome issue then because there were no changes within the last month.
Could you provide more logs? While connected via serial would be best.

moon4-hacs, 2025-07-16:
Still the same with Arduino and IDF gives me this Message after the newest ESP Home Update:

[image]

So this is now effectively broken for me. Cant upload with IDF because of this Error and Arduino keeps disconnecting + the "lock.LOCK_SCHEMA" thing....

moon4-hacs, 2025-07-16:
OK its even worse. Cant upload with Arduino either

[image]

moon4-hacs, 2025-07-16:
I am using ESPHome stable. No Beta or dev.

For Arduino its 1:1 from the "How to Use Page"

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock
 
+

  framework:
    type: arduino

And for IDF its:

external_components:
  - source: github://uriyacovy/ESPHome_nuki_lock@esp-idf

+ 

  framework:
    type: esp-idf

IDF upload still fails with the same error.

moon4-hacs, 2025-07-16:
The initial Error "Please use lock.lock_schema..." is gone. But the compilation still fails --> 

[image]

AzonInc, 2025-07-16:
Okay I found the issue.
They introduced new guard defines to reduce the memory/flash usage to exclude things that are not needed.

Please try again and add that to your config:
```
api:
  custom_services: true
```