uriyacovy/ESPHome_nuki_lock #80: Requestkeyturnerstate has resulted in failed Error

Fannangir, 2025-08-22:
The error has appeared after update of ESP to revision 2025.8.0.

It asked to add homeassistant_services: true to the api section (could be related?)

[image]

Is there a solution?

etzisim, 2025-08-29:
I’m experiencing the same issue.
Setup: ESPHome v2025.8.2, running on esp32-dev.
After unpairi can't pair again
The behavior matches the description above.

AzonInc, 2025-08-31:
See my comment in #81 
That might fix the pairing issue as well but I couldn't check yet.

Fannangir, 2025-08-31:
Tried the solution of changing to new branch.
I get the following:

[image]

Seems like something is missing in this branch.
Should I remove it or the branch should be updated?

AzonInc, 2025-08-31:
Oh yeah sorry, I forgot to mention that this parameter was removed, you can just remove that config line.

AzonInc, 2025-09-17:
This is probably related to the version bump of Arduino and IDF.

At the moment I have no idea what needs to be adjusted in order to make it work again.

Also I'm gonna be on holidays for another two weeks before I can finally take a look.

AzonInc, 2025-10-07:
This might fix it: https://github.com/uriyacovy/ESPHome_nuki_lock/pull/84